openthread/openthread #2350: Commissioning is broke

DuaneEllis-TI, 2017-11-16:
We think this Issue: https://github.com/openthread/openthread/issues/2254
And this merge: https://github.com/openthread/openthread/pull/2311

Is causing some problems with the Thread Conformance/Certification scripts from GRL

Which assumes the previous functionality...

In the end, the commissioning tests fail.

DuaneEllis-TI, 2017-11-16:
Hmm - Update on this - 

The GRL  DUT code for the commissioner tests is given/passed as a parameter the EUI64 - then for the openthread configuration it calculates a the HASH MAC and performs the ADD  using the HASHMAC

The solution is quite simple:  require 2 different Python scripts for openthread based devices.

Version: Original - remains unchanged

Version: NEW - passes the EUI64 not the HASHMAC ... to the DUT.

jwhui, 2017-11-17:
@DuaneEllis-TI, I'm not quite clear on what the issue is, but let me try to answer:

OpenThread reference devices driven by the Thread Test Harness should use the OpenThread python script provided with the Thread Test Harness.

For the OpenThread DUT device:
1) Drive the DUT manually.
2) Add the latest OpenThread.py file from GitHub as a [new reference platform](https://openthread.io/certification/automation_setup#acting_as_a_new_reference_platform).

I believe your "2 different Python scripts" is the latter option above.

Is there anything that we need to change on GitHub to address your issue?

jwhui, 2017-11-27:
@DuaneEllis-TI, were you able to make progress on this issue?

DuaneEllis-TI, 2017-11-27:
Yes we have - sorry should have updated this issue and closed it earlier.

closed 2017-11-27 by DuaneEllis-TI