openthread/openthread #2357: MLE: MTD's mode setting?

Hyungsin, 2017-11-21:
I tested MTD mode, which does not make a node sleep.
Why is "RxOnWhenIdle" turned on here?
https://github.com/openthread/openthread/blob/master/src/core/thread/mle.cpp#L67

GitHub Issues are for bugs and feature requests.  To make bugs and feature requests more easy to find and organize, we close issues that are deemed out of scope for GitHub Issues.

Usage questions? Post questions to [Stack Overflow](http://stackoverflow.com/) using the [[openthread] tag](http://stackoverflow.com/questions/tagged/openthread). We also use Google Groups for discussion and announcements:

* [openthread-announce](https://groups.google.com/forum/#!forum/openthread-announce) - subscribe for release notes and new updates on OpenThread
* [openthread-users](https://groups.google.com/forum/#!forum/openthread-users) - the best place for users to discuss OpenThread and interact with the OpenThread team
* [openthread-devel](https://groups.google.com/forum/#!forum/openthread-devel) - team members discuss the on-going development of OpenThread

jwhui, 2017-11-21:
The radio is always-on when attaching to a network, in order to receive MLE Parent Response messages.  A MTD only receives messages via IEEE 802.15.4 Indirect Transmissions after it has selected a Parent.

Hyungsin, 2017-11-21:
Yes, but to make a node sleep, I think SetRxOnWhenIdle(false) should be executed as follows.
https://github.com/openthread/openthread/blob/master/src/core/thread/mle.cpp#L1680
https://github.com/openthread/openthread/blob/master/src/core/thread/mle.cpp#L1824
https://github.com/openthread/openthread/blob/master/src/core/thread/mle.cpp#L2900

When RxOnWhenIdle is on in mDeviceMode, the above lines are not executed.

jwhui, 2017-11-21:
Minimal Thread Devices (MTDs) may operate as a Minimal End Device (MED) or Sleepy End Device (SED).  The former is the default configuration in OpenThread.  The latter may be configured by setting the rx-on-when-idle flag to false.

Hyungsin, 2017-11-21:
Oh, I did not know that SED is not default. 
Then it's fine if the default setting is documented somewhere since SED is also working well with disabling rx-on-when-idle.

jwhui, 2017-11-22:
See the [`otThreadSetLinkMode`](https://github.com/openthread/openthread/blob/master/include/openthread/thread.h#L211) API.

closed 2017-11-22 by jwhui