openthread/openthread #2358: Routing: Handling the link with cost 4

Hyungsin, 2017-11-21:
I see the following note on the page 5-25 of Thread specification 1.1.1.
[image]

In my understanding, Thread wants to select a neighbor with link cost 4 as the nexthop only if there is no other choice. But the following line makes me feel that OpenThread may prefer a direct transmission to the destination with link cost 4 when better multihop routes exist.
https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder.cpp#L958

jwhui, 2017-11-21:
General consensus is that Link Cost 4 links included in the route selection as other link metrics.  I can see how the specification text is confusing.  It may be worth filing an errata JIRA.  At the same time, the text does not contain normative keywords.

See [SPEC-218](https://threadgroup.atlassian.net/browse/SPEC-218) for more discussion around this topic.

Hyungsin, 2017-11-21:
Thanks! But I cannot access SPEC-218.

jwhui, 2017-11-21:
You need to be a Thread Group member.

Hyungsin, 2017-11-21:
Ok. Is academic membership enough to see this?

Also regarding the following line, does OpenThread want to use direct route even though there are better multihop routes? For example, when the direct route has cost 4 and the two-hop route has cost 2 (1+1), OpenThread seems to select the direct route.
https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder.cpp#L958

Hyungsin, 2017-11-21:
I'm curious about the reason why link cost 4 is not handled differently. Avoiding using link cost 4 links may increase hop distance a little bit, but it would not be problematic since routers are plugged-in and always-on.

jwhui, 2017-11-21:
Yes.  Academic and Affiliate membership have the same benefits: http://threadgroup.org/joinus#Membershipbenefits

In practical settings, going multiple hops incurs a lot more risk than the link/path cost may suggest.  In addition to link margin, there's buffering constraints, routing inconsistencies, etc.  If you can deliver directly to the destination, you avoid all of that uncertainty.

If the direct link turns out to be unusable, the direct link will be removed and the routed path will be used.

Hyungsin, 2017-11-21:
I see, thanks!
Maybe UC Berkeley is not an academic member yet?

jwhui, 2017-11-21:
I don't think so.  I reached out to you earlier in the year... :)

Hyungsin, 2017-11-21:
and we answered "Yes"... :) So does it take a while?

jwhui, 2017-11-21:
You have to apply: http://threadgroup.org/join/Becomemember

Hyungsin, 2017-11-21:
Got it. Thanks!

closed 2017-11-22 by jwhui

Hyungsin, 2017-11-30:
Assume that a neighbor has a weak link which becomes on and off.
Then, the current openthread tries to send packets directly to the neighbor whenever the link is on and experiences continuous failures when the link is off. Is this cycle repeated infinitely without giving any penalty to that neighbor node?