openthread/openthread #2359: Secure coap api in ot? [new feature]

duenzhaw, 2017-11-21:
What is the best way to implement secure coap with ot? It would be possible to implement secure coap on top of the udp api of ot, although ot uses secure coap by itself.
One can find a secure coap implementation in [coap_secure.cpp](https://github.com/openthread/openthread/tree/master/src/core/coap). However, there is no secure coap api.
If a secure coap api is implemented, there must be a possibility to adjust some dtls attributes too: which ciphersuites to use and where to load certificates from...

Which strategy would you recommend? Is there already an implementation of secure coap with ot?

jwhui, 2017-11-27:
@duenzhaw, exposing a secure coap functionality sounds like a great enhancement.  I'm not aware of any existing work that does this already.  The particular strategy to recommend largely depends on the use case.

On the simple end of the spectrum, the use case only requires a basic CoAP implementation and does not need to operate simultaneously with Thread Commissioning roles.  In this case, I would introduce APIs that expose the existing secure coap methods plus minimal set of methods needed for DTLS-specific configuration (certificates, etc.).  Of course, the `mbedtls-config.h` file would also need to be updated to include the necessary cipher suites are part of the build.  If you need additional CoAP functionality, we could also look to enhancing OpenThread's CoAP implementation.

On the more flexible end of the spectrum, you could utilize the mbedtls library on top of OpenThread's UDP API.  This gives you full access to the mbedtls APIs and full flexibility on memory utilization.  This would also allow you to provide a more full-featured CoAP implementation on top.  However, this would probably require more implementation effort in the near term.

Thoughts?  Are you interested in submitting a PR once we have agreed on a path forward?

duenzhaw, 2017-12-12:
We are still doing conceptional work. As we only require basic CoAP implementation (until now), everything indicates that we start on the simple end of spectrum: implementation of CoAPS api within OT (not on top of the udp api).
We have started a list of changes required for adding such an api:
- mbedtls-config.h
  - enable ciphersuites (We probably use: mbedtls: TLS-ECDHE-ECDSA-WITH-AES-128-CCM-8, value: {0xC0,0xAE})
  - x509 cert
  - ...
- coap.*
  -  add block support
- coap C wrapper
- coaps.*
- create coaps C wrapper
- dtls.*
  - certificates trough coaps api?

What do you think?

ChrIgiSta, 2018-05-16:
We have just made a PR: #2716 

Regards