openthread/openthread #2360: [CC2650 example][Building error] Dereferencing type-punned pointer will break strict-aliasing rules

TobiasReichert, 2017-11-21:
I have tried to build the CC2650 example.

Steps i made:
```bash
$ cd <path-to-openthread>
$ ./script/bootstrap
$ ./bootstrap
$ make -f examples/Makefile-cc2650
```

But wenn i do the last one i get this error 
```
[...]/openthread/examples/../src/core/common/instance.cpp: In static member function 'static ot::Instance& ot::Instance::Get()':
[...]/openthread/examples/../src/core/common/instance.cpp:96:55: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     return *reinterpret_cast<Instance *>(&sInstanceRaw);

```

This guy seems to have the same error: https://github.com/openthread/openthread/pull/2332#issuecomment-345943567

arm-none-eabi-gcc version:
```
gcc version 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437] (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 
```
Hope you can help =)

jwhui, 2017-11-22:
Proposed fix in #2361.

TobiasReichert, 2017-11-22:
Fixed by:
https://github.com/openthread/openthread/pull/2361

Thank you =)

PR #2361 (2017-11-22): [instance] fix strict-aliasing warning for gcc6
Resolves #2360.
closed 2017-11-22 by TobiasReichert