openthread/openthread #2364: REED and FED receiving of multicasts from neighboring routers

CiaranWoodward, 2017-11-22:
According to section 4.7.7.4 of thread spec 1.1.1 REEDs and FEDs must maintain one-way links with neighboring routers so that the REED/FED can receive MAC-secured multicasts from them. Currently in openthread FEDs/REEDs use link requests & accepts as required, but drop incoming MAC-secured transmissions that are not from their parents.

This is because the method of validating a MAC source address uses GetNeighbor(), which switches on the current role:
https://github.com/openthread/openthread/blob/1f6528e6c9a600b7f04128ed8f22e841107361f6/src/core/mac/mac.cpp#L1572-L1576
Of course, this behaviour of GetNeighbor is usually needed, so perhaps a special case in Mac.cpp to check the router table too, in the case of an FFD child receiving a broadcast frame?

Example Logs (I used an address query from a non-parent router to generate):
Child establishing one way link:
```
[6] [INFO]-MLE-----: Receive Advertisement (fe80:0:0:0:2c18:59bb:9377:9a0f,0x8000)
[6] [INFO]-MLE-----: Send Link Request (fe80:0:0:0:2c18:59bb:9377:9a0f)
[6] [INFO]-MAC-----: Sent IPv6 UDP msg, len:92, chksum:1cb5, to:2e1859bb93779a0f, sec:no, prio:high
[6] [INFO]-MAC-----: src: fe80:0:0:0:1432:68b9:eb2d:53b7
[6] [INFO]-MAC-----: dst: fe80:0:0:0:2c18:59bb:9377:9a0f
[6] [INFO]-MAC-----: Received IPv6 UDP msg, len:97, chksum:8e81, from:2e1859bb93779a0f, sec:no, prio:low, rss:-20.0
[6] [INFO]-MAC-----: src: fe80:0:0:0:2c18:59bb:9377:9a0f
[6] [INFO]-MAC-----: dst: fe80:0:0:0:1432:68b9:eb2d:53b7
[6] [INFO]-MLE-----: Receive Link Accept (fe80:0:0:0:2c18:59bb:9377:9a0f,0x8000)
```

Router table of child:
```
router table[6] [INFO]---------: execute command: router table

| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
| 29 | 0x7400 |       29 |         1 |     0 |      0 |  41 | 0000000000000000 |
| 32 | 0x8000 |        0 |         0 |     3 |      3 |   4 | 2e1859bb93779a0f |

Done
```

And the filtering upon receiving of a multicast:
```
[6] [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:65, seqnum:195, type:Data, src:0x8000, dst:0xffff, sec:yes, ackreq:no
[6] [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:65, seqnum:196, type:Data, src:0x8000, dst:0xffff, sec:yes, ackreq:no
```

jwhui, 2017-11-27:
@CiaranWoodward, good catch!  Would you like to submit a PR with the proposed fix?

CiaranWoodward, 2017-11-27:
@jwhui Sure, I'll look further into it and submit a PR soon

PR #2373 (2017-12-04): [mac] process security for neighbors on FFD Children
Fixes #2364 

Also simplified the existing MleRouter::GetNeighbor() slightly
closed 2017-12-04 by jwhui