openthread/openthread #2372: STATUS_NOMEM on SiLabs EFR32

tomas-c, 2017-11-28:
Hi there,

I've been testing SiLabs MGM12P modules with OpenThread. After a 30min to 2 hours of operation the NCP stops transmitting/receiving IPv6 packets. On every transmit wpantund tells me that NCPs last status is STATUS_NOMEM.

```
Nov 27 23:35:49  wpantund[399]: Tickle...
Nov 27 23:35:49  wpantund[399]: [->NCP] CMD_NOOP tid:7
Nov 27 23:35:49  wpantund[399]: [NCP->] CMD_PROP_VALUE_IS(PROP_LAST_STATUS) tid:7
Nov 27 23:35:49  wpantund[399]: [-NCP-]: Last status (STATUS_OK, 0)
Nov 27 23:35:51  wpantund[399]: [->NCP] IPv6 len:115 type:17(cksum 0xa257) [SECURE]
Nov 27 23:35:51  wpantund[399]:         to(remote):[ff03::1]:4000
Nov 27 23:35:51  wpantund[399]:         from(local):[fd41:4d42:4552:0:4ee:14d0:6710:a5c5]:56943
Nov 27 23:35:56  wpantund[399]: [->NCP] IPv6 len:115 type:17(cksum 0x19f2) [SECURE]
Nov 27 23:35:56  wpantund[399]:         to(remote):[ff03::1]:4000
Nov 27 23:35:56  wpantund[399]:         from(local):[fd41:4d42:4552:0:4ee:14d0:6710:a5c5]:56943
Nov 27 23:35:56  wpantund[399]: [NCP->] CMD_PROP_VALUE_IS(PROP_LAST_STATUS) tid:0
Nov 27 23:35:56  wpantund[399]: [-NCP-]: Last status (STATUS_NOMEM, 11)
Nov 27 23:36:01  wpantund[399]: [->NCP] IPv6 len:115 type:17(cksum 0xda92) [SECURE]
```

Has anyone else encountered this before?

What's the best way to debug this when it takes several hours to reproduce?

abtink, 2017-11-28:
@tomas-c  the `NOMEM` typically indicates that that the OT message buffers may be all in use (so `NcpBase` cannot allocate a `Message` for a new IP message). We have seen this symptom in past and the root cause can be different things. 
Here are some suggestions for further debugging: 
- You can get the status of the OT buffer using the wpantund property `"OpenThread:MsgBufferCounters"` when device is in this state. It provides info about total number of buffers, and which module is holding on to them (please see https://github.com/openthread/openthread/blob/master/src/ncp/spinel.h#L1516 for the spinel property and different counters)
-  _This symptom is typically caused by something in platform layer not invoking a callback_. 
   - For example, we have seen this  in a case where the underlying radio platform could (rarely) miss a send-done callback.  OpenThread MAC implementation requires the underlying radio to always invoke a send-done after issuing a send,  otherwise it would indefinitely wait for it. This can cause the incoming messages to fill the send queue running out of buffer.
   - You can consider enabling debug level logs which would then provide details about state of `Mac` (when an operation (e.g. tx) starts and ends).

tomas-c, 2017-11-29:
@abtink Thanks for the suggestions.

I tried enabling NCP debug logs that but seems to use up all the UART bandwidth which results in almost immediate STATE_NOMEM.

As for OpenThread:MsgBufferCounters, I get the following;

```
OpenThread:MsgBufferCounters = [
	"mTotalBuffers = 40"
	"mFreeBuffers = 0"
	"m6loSendMessages = 14"
	"m6loSendBuffers = 40"
	"m6loReassemblyMessages = 0"
	"m6loReassemblyBuffers = 0"
	"mIp6Messages = 0"
	"mIp6Buffers = 0"
	"mMplMessages = 0"
	"mMplBuffers = 0"
	"mMleMessages = 0"
	"mMleBuffers = 0"
	"mArpMessages = 0"
	"mArpBuffers = 0"
	"mCoapClientMessages = 0"
	"mCoapClientBuffers = 0"
]
```

mFreeBuffers is 0 but total number of messages is less than mTotalBuffers. Is this normal or am I missing something?

abtink, 2017-11-29:
From the counters it looks like all buffers are being used in the "send queue".

An `ot::Messages` (which can contain a full IPv6 payload) can itself contain multiple data buffers (which are fixed size). The counter numbers are ok. There are `"m6loSendMessages = 14"` messages in the send queue which consume total of  `"m6loSendBuffers = 40"` (which is the total number of buffers).

The likely case I suggest to investigate is for radio platform missing (not invoking) send-done callback after tx (mac layer would then be waiting for a radio platform callback and all new messages pill up in send queue). Maybe trying to add selective debug logs in `Mac` implementation to check for this.

tomas-c, 2017-11-30:
Was able to increase serial baud rate to ~1MHz and that was enough to avoid MAC log messages interfering with normal operation.

I have attached a log file of one of the devices that experiences the issue. A bit difficult to read as wpantund doesn't include new lines when printing NCP log messages.

I did notice one thing. At around 14:03:20 the MAC layer prints "Starting operation" twice without "Finishing" in between and then never finishes after. Shortly after NOMEM shows up.

NCP MAC log at around Nov 30 14:03:20:
```
[DEBG]-MAC-----: Finishing operation "TransmitData"
[DEBG]-MAC-----: Idle mode: Radio receiving on channel 14
[DEBG]-MAC-----: Received frame from short address 0x9800
[DEBG]-MAC-----: Frame counter 99
[INFO]-MAC-----: Received IPv6 HopOpts msg, len:123, chksum:0000, from:0x9800, sec:yes, prio:low, rss:-54.0
[INFO]-MAC-----: src: fd41:4d42:4552:0:a6c2:31c1:2976:20c7
[INFO]-MAC-----: dst: ff03:0:0:0:0:0:0:1
[DEBG]-MAC-----: Request to start operation "TransmitData"
[DEBG]-MAC-----: Starting operation "TransmitData"
[INFO]-MAC-----: Frame tx failed, error:NoAck, attempt:1/4, len:125, seqnum:75, type:Data, src:0x4000, dst:0x2c00, sec:yes, ack
[INFO]-MAC-----: Frame tx failed, error:NoAck, attempt:2/4, len:125, seqnum:75, type:Data, src:0x4000, dst:0x2c00, sec:yes, ack
[INFO]-MAC-----: Frame tx failed, error:NoAck, attempt:3/4, len:125, seqnum:75, type:Data, src:0x4000, dst:0x2c00, sec:yes, ack
[INFO]-MAC-----: Frame tx failed, error:NoAck, attempt:4/4, len:125, seqnum:75, type:Data, src:0x4000, dst:0x2c00, sec:yes, ack
[INFO]-MAC-----: Failed to send IPv6 UDP msg, len:130, chksum:4a27, to:0x2c00, sec:yes, error:NoAck, prio:low
[INFO]-MAC-----: src: fd41:4d42:4552:0:d790:17b3:3434:cd3c
[INFO]-MAC-----: dst: fd41:4d42:4552:0:a6c2:31c1:2976:20c7
[DEBG]-MAC-----: Finishing operation "TransmitData"
[DEBG]-MAC-----: Idle mode: Radio receiving on channel 14
[INFO]-MAC-----: Received IPv6 UDP msg, len:94, chksum:2ffd, from:fa73e918d7b1e309, sec:no, prio:low, rss:-43.0
[INFO]-MAC-----: src: fe80:0:0:0:f873:e918:d7b1:e309
[INFO]-MAC-----: dst: ff02:0:0:0:0:0:0:1
[DEBG]-MAC-----: Request to start operation "TransmitData"
[DEBG]-MAC-----: Starting operation "TransmitData"
[INFO]-MAC-----: Sent IPv6 UDP msg, len:94, chksum:ed48, to:0xffff, sec:no, prio:high
[INFO]-MAC-----: src: fe80:0:0:0:64fd:58ea:3f95:25a6
[INFO]-MAC-----: Received IPv6 UDP msg, len:94, chksum:3077, from:963fc7ef9802ba53, sec:no, prio:low, rss:-56.0
[INFO]-MAC-----: src: fe80:0:0:0:943f:c7ef:9802:ba53
[INFO]-MAC-----: dst: ff02:0:0:0:0:0:0:1
[DEBG]-MAC-----: Request to start operation "TransmitData"
[DEBG]-MAC-----: Starting operation "TransmitData"
[INFO]-MAC-----: Received IPv6 UDP msg, len:94, chksum:b5bb, from:7e3bfcf3bcab7732, sec:no, prio:low, rss:-53.0
[INFO]-MAC-----: src: fe80:0:0:0:7c3b:fcf3:bcab:7732
[INFO]-MAC-----: dst: ff02:0:0:0:0:0:0:1
[INFO]-MAC-----: Received IPv6 UDP msg, len:94, chksum:b37d, from:fa73e918d7b1e309, sec:no, prio:low, rss:-44.0
[INFO]-MAC-----: src: fe80:0:0:0:f873:e918:d7b1:e309
[INFO]-MAC-----: dst: ff02:0:0:0:0:0:0:1
```

Full wpantund logs:
[status_nomem_log_with_mac_debug_enabled.txt](https://github.com/openthread/openthread/files/1518145/status_nomem_log_with_mac_debug_enabled.txt)

jwhui, 2017-11-30:
From the logs, it appears as we expected - the radio is getting stuck on a transmit operation for some reason.

Would it be possible for you to add a log just before the call to `otPlatRadioTransmit()` and another in `otPlatRadioTxDone()` to see if the callback is getting called properly?

abtink, 2017-11-30:
Exactly...Seems like radio platform code may be getting stuck on a tx and not invoking callback.

For to code to add `\n` to all OT log and make the logs easier to read, you can change this config option `OPENTHREAD_CONFIG_LOG_SUFFIX` and set it to be `"\n"`....Then OT log functions would then automatically add the new line.

tomas-c, 2017-12-04:
Thanks, guys. I added the log messages and jwhui suggested. Logs attached. 

It works for a while but then at 14:11:28, after the last "otPlatRadioTransmit" call, "otPlatRadioTxDone" is never called back.

I thought maybe RAIL_TxDataLoad or RAIL_TxStartWithOptions calls inside otPlatRadioTransmit fail but I ran another test and both of those always return 0, even on the last call to otPlatRadioTransmit at 14:11:28.

[status_nomem_log_with_mac_debug_enabled.txt](https://github.com/openthread/openthread/files/1529010/status_nomem_log_with_mac_debug_enabled.txt)

tomas-c, 2017-12-04:
Just a guess but the logic in efr32RadioProcess for calling back otPlatRadioTxDone doesn't seem to handle the case when the correct ACK is missed and an ACK for a different transmission is received.

tomas-c, 2017-12-05:
Looks like that's exactly what happens. It gets an acknowledgement but DSN doesn't match so keeps waiting forever.

jwhui, 2017-12-06:
@tomas-c, this issue may be addressed in #2401.

tomas-c, 2017-12-21:
@jwhui, just tested - not an issue with #2401

tomas-c, 2017-12-26:
@jwhui Was a bit too eager to speak - still an issue with Rail 2.0 update.

After 5h of operation one of four devices I had in a network encountered NOMEM.

jwhui, 2018-01-03:
@tomas-c, thanks for the feedback.  Is the observed behavior the same as what you've described above?  In particular, is the `otPlatRadioTxDone()` not coming back?

tomas-c, 2018-01-04:
The behaviour was different in my last test. The the stack didn't get permanently stuck in NOMEM state anymore but NOMEM did happen quite often and resulted in messages being lost.

I have attached logs from my latest test. I added some more log messages to track when radio starts transmitting and when finishes (see attached radio.c).

`State=OT_RADIO_STATE_TRANSMIT, DSN=%d, ackReq=%d` - logged inside otPlatRadioTransmit after otEXPECT_ACTION

`NCP => [INFO]-PLAT----: Transmit done, DSN=%d` - logged inside efr32RadioProcess just before invoking otPlatRadioTxDone

Check out time "Jan 04 05:22:57". otPlatRadioTransmit is called ("State=OT_RADIO_STATE_TRANSMIT, DSN=178, ackReq=0") but the next otPlatRadioTxDone is for a different transmission ("Transmit done, DSN=182") at "Jan 04 05:23:00".
 
[intermittent_nomem_logs_jan03.txt.zip](https://github.com/openthread/openthread/files/1603297/intermittent_nomem_logs_jan03.txt.zip)

[radio_with_more_logging.c.txt](https://github.com/openthread/openthread/files/1603333/radio_with_more_logging.c.txt)

jwhui, 2018-01-04:
@tomas-c, looking at the log, it seems that each `NOMEM` occurs shortly after three UDP multicast messages (ff03::1 destination) are sent back-to-back from wpantund to the NCP.  There are many other occurrences of a single multicast message being sent from wpantund to NCP, but those do not cause `NOMEM` indications.

One example:
```
Jan 04 05:41:20 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:20 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:20 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:20 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:21 -gateway- wpantund[293]: [-NCP-]: Last status (STATUS_NOMEM, 11)
Jan 04 05:41:21 -gateway- wpantund[293]: [-NCP-]: Last status (STATUS_NOMEM, 11)
Jan 04 05:41:21 -gateway- wpantund[293]: [-NCP-]: Last status (STATUS_NOMEM, 11)
Jan 04 05:41:21 -gateway- wpantund[293]: [-NCP-]: Last status (STATUS_NOMEM, 11)
Jan 04 05:41:23 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:29 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:34 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:39 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:44 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:49 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:54 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:41:59 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:42:04 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:42:09 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:42:14 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:42:19 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
Jan 04 05:42:24 -gateway- wpantund[293]:         to(remote):[ff03::1]:4000
```

Can you describe more the traffic pattern that you are testing with?  It would be nice to understand if the traffic load is within what the NCP (and the configured message buffer allocation) should be able to handle.

tomas-c, 2018-01-08:
@jwhui I have four devices. One device (call it "server") transmits a beacon to ff03::1 (single CoAP, non-confirmable multicast) every 5 seconds. The other three devices (call them "clients") upon receiving this beacon start sending to the "server" heartbeart messages every 5 seconds (single CoAP, non-confirmable unicast). In the test, the devices are at max transmission power and within 5m of each other.

I finally got the system to run for 36h without any issues, with

- Using latest code from from #2401
- 115200 baud rate
- hardware flow control
- all NCP logging disabled
- OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE increased to 4096
- OPENTHREAD_CONFIG_NCP_UART_TX_CHUNK_SIZE increased to 512
- uart.c modified to use a 256 byte ring buffer being serviced by two UARTDRV_Receive() calls in 32 byte chunks

It's difficult to tell at this point which of these were necessary (will not have time to investigate for a while) but if it was just a matter of having 115200 baud rate, flow control and no logging then once #2401 is merged - this should be solved.

PR #2401 (2018-01-09): [efr32] update to RAIL 2.x
This update also makes the following changes:
- Initialize chip using `halInitChipSpecific()` provided by RAIL HAL.
- Use `RAIL_HoldRxPacket()` to move rx processing out of interrupt context
- Eliminate critical sections from `otPlatRadio*` APIs.
- Use UART driver provided by Gecko SDK.

Resolves #2372
Resolves #2379

//cc @tomas-c @plasticchris @xiaom-GitHub
closed 2018-01-09 by jwhui