openthread/openthread #2379: efr32 CSMA done twice?

tomas-c, 2017-11-30:
It seems that both the radio and OpenThread perform CSMA.

In efr32 radio.c function otPlatRadioTransmit
```
RAIL_CsmaConfig_t csmaConfig = RAIL_CSMA_CONFIG_802_15_4_2003_2p4_GHz_OQPSK_CSMA;
...
railTxStatus = RAIL_TxStartWithOptions(aFrame->mChannel, &txOption, RAIL_CcaCsma, &csmaConfig);
```

RAIL_CSMA_CONFIG_802_15_4_2003_2p4_GHz_OQPSK_CSMA is

```
/**
 * @def RAIL_CSMA_CONFIG_802_15_4_2003_2p4_GHz_OQPSK_CSMA
 * @brief RAIL_CsmaConfig_t initializer configuring CSMA per 802.15.4-2003
 * on 2.4 GHz OSPSK, commonly used by ZigBee.
 */
#define RAIL_CSMA_CONFIG_802_15_4_2003_2p4_GHz_OQPSK_CSMA {                    \
    /* CSMA per 802.15.4-2003 on 2.4 GHz OSPSK, commonly used by ZigBee     */ \
    /* csmaMinBoExp */ 3,   /* 2^3-1 for 0..7 backoffs on 1st try           */ \
    /* csmaMaxBoExp */ 5,   /* 2^5-1 for 0..31 backoffs on 3rd+ tries       */ \
    /* csmaTries    */ 5,   /* 5 tries overall (4 re-tries)                 */ \
    /* ccaThreshold */ -75, /* 10 dB above sensitivity                      */ \
    /* ccaBackoff   */ 320, /* 20 symbols at 16 us/symbol                   */ \
    /* ccaDuration  */ 128, /* 8 symbols at 16 us/symbol                    */ \
    /* csmaTimeout  */ 0,   /* no timeout                                   */ \
}
```

so SiLabs Rail is set to use CSMA.

Now if we look into MAC layer, we see that before transmitting it calls `RadioSupportsCsmaBackoff()` and if it returns false - performs its own CSMA. In efr32 radio.c function, otPlatRadioGetCaps doesn't set OT_RADIO_CAPS_TRANSMIT_RETRIES or OT_RADIO_CAPS_CSMA_BACKOFF bits so this is precisely what happens.

xiaom-GitHub, 2017-12-01:
Thanks for helping test and debug the root cause！sorry to bring this issue. Just as you mentioned, please supplement OT_RADIO_CAPS_CSMA_BACKOFF & OT_RADIO_CAPS_TRANSMIT_RETRIES cap in otPlatRadioGetCaps() to see if the issue will be fixed. If so, would you mind to submit a PR for this bug? Thanks!

tomas-c, 2017-12-04:
Thanks, Xiaom. Will do once I get #2372 sorted. Thought just mention now before someone else starts looking.

jwhui, 2017-12-06:
I've addressed this issue as part of the transition to RAIL 2.x in #2401.

PR #2401 (2018-01-09): [efr32] update to RAIL 2.x
This update also makes the following changes:
- Initialize chip using `halInitChipSpecific()` provided by RAIL HAL.
- Use `RAIL_HoldRxPacket()` to move rx processing out of interrupt context
- Eliminate critical sections from `otPlatRadio*` APIs.
- Use UART driver provided by Gecko SDK.

Resolves #2372
Resolves #2379

//cc @tomas-c @plasticchris @xiaom-GitHub
closed 2018-01-09 by jwhui