openthread/openthread #2380: nrf52480: Calling `nrf_drv_radio802154_fsm_sleep() breaks usb-cdc-uart transmit

darconeous, 2017-11-30:
I am currently working on bringing up a nRF52480-based NCP that uses its native USB capabilities to communicate with the application/host processor, using [USB-CDC-ACM UART implementation](https://github.com/openthread/openthread/blob/master/examples/platforms/nrf52840/usb-cdc-uart.c) (documented [here](https://github.com/openthread/openthread/tree/master/examples/platforms/nrf52840#native-usb-support)).

At first it seemed like it was working fairly well, but then I noticed that the NCP would become unresponsive after receiving certain commands. Depending on the command there were a few different reasons for this, but this bug is about the freeze that would happen after doing something like setting `PROP_NET_STACK_UP` to false.

When I would set `PROP_NET_STACK_UP` to false, I would no longer receive any output on the USB-CDC-ACM port. However, I know that the commands I was sending it were being received and processed.

Looking into the issue further made me realize that `ncp_uart.cpp` had called `otPlatUartSend()` to send the host the next frame, but `otPlatUartSendDone()` was not being called. Digging into this further, I determined that one of the results of setting `PROP_NET_STACK_UP` to false is a call to `otPlatRadioSleep()`. When I modified this function to return without doing anything, setting `PROP_NET_STACK_UP` to false would succeed.

Digging even further, I narrowed down the problem to the call to `nrf_drv_radio802154_fsm_sleep()` (Called from `nrf_drv_radio802154_request_sleep()`, which is called from `nrf_drv_radio802154_sleep()`, which is called from `otPlatRadioSleep()`). Calling `nrf_drv_radio802154_fsm_sleep()` somehow seems to quite reliably prevent future calls to `app_usbd_cdc_acm_write()` from triggering `APP_USBD_CDC_ACM_USER_EVT_TX_DONE`. It does not, however, prevent `APP_USBD_CDC_ACM_USER_EVT_RX_DONE` from being triggered, as even in this state the NCP continues to process inbound frames.

darconeous, 2017-11-30:
It seems like things go off the rails once `sleep_start()` in `nrf_drv_radio802154_fsm.c` is called as a result of calling `otThreadSetEnabled(false)` (when it had previously been enabled).

darconeous, 2017-11-30:
Poking around some more, it seems that both the RAAL and USBD use HFCLK. Neutering the macro `NRF_RAAL_HFCLK_STOP()` prevents this problem from occurring.

darconeous, 2017-11-30:
Of course, it's never that simple, as after that I'm seeing issues when attempting to scan. But hopefully that is enough information to help someone more familiar with the code to figure out what is ultimately wrong and the best way to fix it.

darconeous, 2017-12-01:
Maybe `nrf_raal_continuous_mode_enter()`/`nrf_raal_continuous_mode_exit()` should be updated to use `nrf_drv_clock`?

Also, Everything seems to lock up pretty badly (no longer handled incoming commands) after a call to `otPlatRadioReceive()` when using USB-CDC-ACM.

rlubos, 2017-12-01:
Thank you for your input, practically you've found the problem yourself.

Indeed for some reason we do not use `nrf_drv_clock` in standalone `nrf_raal`. I'll propose a fix soon.

PR #2381 (2017-12-03): [nrf52840] Fix HF clock stop procedure in radio driver
Should fix #2380 . I don't have NCP USB setup to test it, but I've verified that now radio driver uses clock driver to enable and disable HF clock.
closed 2017-12-03 by jwhui