openthread/openthread #2382: [nrf52480] usb-cdc-uart: Sending/receiving beacons locks up NCP

darconeous, 2017-12-01:
It would appear that when used in usb-cdc-uart mode that receiving or sending a beacon locks up the NCP hard (as opposed to #2380 where only usb-cdc-uart transmit was jammed).

Responding to a beacon request:

```
[0000068246] [INFO]-MAC-----: Received Beacon Request
[0000068246] [DEBG]-MAC-----: Request to start operation "TransmitBeacon"
[0000068247] otPlatRadioReceive
[0000068247] [DEBG]-MAC-----: Starting operation "TransmitBeacon"
[0000068249] [INFO]-MAC-----: Sending Beacon, name:LoWPAN_352, xpanid:87ecc1eb76b5d7c0, id:3 ver:2, joinable:no, ? 
***LOCKED UP***
```

Performing a scan where we received a beacon response:

```
[0000039579] [DEBG]-MAC-----: Request to start operation "ActiveScan"
[0000039579] otPlatRadioReceive
[0000039579] [DEBG]-MAC-----: Starting operation "ActiveScan"
[0000039579] otPlatUartSend: 8 bytes
[0000039580] APP_USBD_CDC_ACM_USER_EVT_TX_DONE
[0000039580] otPlatUartSendDone
[0000039581] [INFO]-MAC-----: Sending Beacon Request
[0000039581] otPlatRadioReceive
[0000039581] otPlatRadioTransmit
[0000039587] [INFO]-MAC-----: Received Beacon, name:blahnet, xpanid:b0a30278d86d2d8d, id:3 ver:2, joinable:no, na?
***LOCKED UP***
```

It is unclear to me if this is beacon-specific, or if this represents a general problem with sending/receiving packets.

Note that this behavior only seems to occur when using `USB=1`. When using the plain vanilla UART I don't see these issues.

darconeous, 2017-12-01:
Looks like some stack corruption has led to the NCP getting stuck in `HardFault_Handler()`:

```
Remote debugging using localhost:2331
HardFault_Handler () at /usr/local/google/home/rquattle/Projects/openthread/examples/../examples/platforms/nrf52840/../../../third_party/NordicSemiconductor/device/gcc_startup_nrf52840.S:428
428                                                       b       .
(gdb) bt
#0  HardFault_Handler () at /usr/local/google/home/rquattle/Projects/openthread/examples/../examples/platforms/nrf52840/../../../third_party/NordicSemiconductor/device/gcc_startup_nrf52840.S:428
#1  <signal handler called>
#2  ot::Mac::Mac::ConvertBeaconToActiveScanResult (this=0x0 <__isr_vector>, aBeaconFrame=0x0 <__isr_vector>, aResult=...) at /usr/local/google/home/rquattle/Projects/openthread/examples/../src/core/mac/mac.cpp:311
#3  0x00000000 in ?? ()
(gdb) up 2
#2  ot::Mac::Mac::ConvertBeaconToActiveScanResult (this=0x0 <__isr_vector>, aBeaconFrame=0x0 <__isr_vector>, aResult=...) at /usr/local/google/home/rquattle/Projects/openthread/examples/../src/core/mac/mac.cpp:311
311         return error;
```

darconeous, 2017-12-01:
It looks like there is a problem with the logging code in `examples/platforms/nrf52840/logging.c` starting on 168:

    length += vsnprintf(&logString[length], (LOG_PARSE_BUFFER_SIZE - length),
                        aFormat, paramList);
    logString[length++] = '\n';
    va_end(paramList);

This code is making an incorrect assumption about the return value of `vsnprintf()`. The return value of the `printf()` functions is **NOT** the number of bytes written to the string, it is the number of characters that it would have output if there were no length constraints. For example, both of the following calls to `snprintf()` will return the value 10:

* `snprintf(mystring, 5, "0123456789");`
* `snprintf(mystring, 20, "0123456789");`

By changing the code in question to the following, the problem is addressed:

    length += vsnprintf(&logString[length], (LOG_PARSE_BUFFER_SIZE - length),
                        aFormat, paramList);
    if (length > LOG_PARSE_BUFFER_SIZE)
    {
        length = LOG_PARSE_BUFFER_SIZE;
    }
    logString[length++] = '\n';
    va_end(paramList);

darconeous, 2017-12-01:
Proposed fix in #2384.

LuDuda, 2017-12-02:
Great catch! Thank you for helping us with this issue. Indeed vsnprintf return value was incorrectly interpreted.

jwhui, 2017-12-04:
Resolved by #2384.

closed 2017-12-04 by jwhui