openthread/openthread #2414: Cannot open Visual Studio .sln project file

VictorAtPL, 2017-12-09:
Hello,

I cloned repository code of OpenThread and tried to open .sln project file located at etc/visual-studio as openthread.sln and I get following error:
[image]

Version of VS: 
[image]

I appreciate any help.

jwhui, 2017-12-12:
Thanks for raising this issue.  I'm not sure why AppVeyor does not run into this specific error.

The solution file was last updated by #2329.  Can you try removing the commented-out lines completely to see if that helps?

VictorAtPL, 2017-12-25:
Yes, it helped.

Could you tell me why did you comment those lines?

jwhui, 2018-01-03:
@VictorAtPL, #2325 details why those lines were commented.  I can submit a PR that directly removes those lines.

jwhui, 2018-01-03:
@VictorAtPL, submitted #2448.  Please review.

PR #2448 (2018-01-08): [appveyor] remove OpenThread project due to expired certificate
Visual Studio does not like commented lines in sln files.  This PR simply
removes the commented lines.

Resolves #2414.
closed 2018-01-08 by jwhui