openthread/openthread #2416: 'cc1plus': execvp: No such file or directory

TobiasReichert, 2017-12-12:
I'm trying to get the CC2650 Example to run.

I have installed the gcc 5.4.1 like it's described in this issue:https://github.com/openthread/openthread/issues/1915#issuecomment-334857701
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

But when i trying to compile this error appears:
```
$ make -f examples/Makefile-cc2650
  BUILD    cc2650
make -j10 -C build/cc2650 --no-print-directory all
  CHECK    .local-version
make --no-print-directory all-recursive
Making all in include
make  all-recursive
Making all in openthread
Making all in platform
make[6]: Für das Ziel „all“ ist nichts zu tun.
make[6]: Für das Ziel „all-am“ ist nichts zu tun.
Making all in src
Making all in core
  CXX      api/libopenthread_ftd_a-coap_api.o
  CXX      api/libopenthread_ftd_a-commissioner_api.o
  CXX      api/libopenthread_ftd_a-child_supervision_api.o
  CXX      api/libopenthread_ftd_a-dataset_api.o
  CXX      api/libopenthread_ftd_a-crypto_api.o
  CXX      api/libopenthread_ftd_a-dhcp6_api.o
  CXX      api/libopenthread_ftd_a-dataset_ftd_api.o
  CXX      api/libopenthread_ftd_a-icmp6_api.o
arm-none-eabi-g++: error trying to exec 'cc1plus': execvp: No such file or directory
arm-none-eabi-g++: error trying to exec 'cc1plus': execvp: No such file or directory
arm-none-eabi-g++: error trying to exec 'cc1plus': execvp: No such file or directory
arm-none-eabi-g++: error trying to exec 'cc1plus': execvp: No such file or directory
Makefile:1814: die Regel für Ziel „api/libopenthread_ftd_a-coap_api.o“ scheiterte
arm-none-eabi-g++: error trying to exec 'cc1plus': make[4]: *** [api/libopenthread_ftd_a-coap_api.o] Fehler 1
arm-none-eabi-g++: error trying to exec 'cc1plus': execvp: No such file or directory
  CXX      api/libopenthread_ftd_a-dns_api.o
make[4]: *** Auf noch nicht beendete Prozesse wird gewartet …
```

What am I doing wrong?

jwhui, 2017-12-12:
It appears your system is 64-bit - you need to install the 32-bit libraries.  See how [here](https://ioprog.com/2015/06/27/setting-up-arm-none-eabi-gcc-on-ubuntu-15-04/).

jwhui, 2017-12-12:
What host OS are you using?  If ubuntu, can you try running:
```bash
$ cd <path-to-openthread>
$ ./script/bootstrap
$ git clean -xfd
$ ./bootstrap
$ make -f examples/Makefile-cc2650
```

TobiasReichert, 2017-12-12:
Oh Sorry.
Yes i'm on Ubuntu.
But if you start the bootstrap it will install the newest arm-gcc, which is on version 6.
In Version 6 there will be this error: https://github.com/openthread/openthread/issues/1915#issue-237021114

So i run the bootstrap, uninstalled the arm-gcc and downloaded the version 5 https://launchpad.net/gcc-arm-embedded/5.0

jwhui, 2017-12-12:
FWIW, the gcc6 errors should be addressed by #2366.  I just built the latest master with no problems using the following version:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

TobiasReichert, 2017-12-12:
Oh okay i will check that.

TobiasReichert, 2017-12-13:
Okay no errors on building with version 6.3.1.
Thank you for the good Support. =)

PS: Maybe it would be good if you change the version hint:
https://github.com/openthread/openthread/blob/master/examples/platforms/cc2650/README.md

closed 2017-12-13 by TobiasReichert