openthread/openthread #2418: Invoking EnergyScan while ActiveScan is ongoing causes crash

abtink, 2017-12-13:
The current `Mac` implementation uses a shared union member variable to store the handler for either active and energy scan. Wit the fix from issue https://github.com/openthread/openthread/issues/2257 (and PR: https://github.com/openthread/openthread/pull/2274) the `mActiveScanHandler` or `mEnergyScanHandler` are set to `NULL` in case of any error in starting the scan operation. This can overwrite an ongoing (previously started) scan handler and causes a `NULL`.

abtink, 2017-12-13:
How about if we use a model similar to `Mac::Sender` and `Mac::Receiver` for active/energy scan operations. Basically adding a new `ScanClient` class which contains the callback logic and other scan parameters. With this model, we can have multiple clients start different scans at the same time and they will be served in the order they are issued. 

Thoughts? @jwhui?

abtink, 2017-12-14:
https://github.com/openthread/openthread/pull/2420 should address this issue.

jwhui, 2017-12-16:
Resolved by #2420.

closed 2017-12-16 by jwhui