openthread/openthread #2425: [tests] unstable tests collection

librasungirl, 2017-12-18:
OT Contributors may have experienced travis tests failure which may be not relative to their PRs. Re-run travis build is current workaround. Now we have dedicated person who will 
help to improve this. If you have met or are experiencing travis tests failure and you think it should not be a problem caused by your PR, please help to list the unstable tests here. Thanks.

abtink, 2017-12-18:
Great to see this. Thanks.
I have seen the following tests failing for me on some travis runs: 
- FAIL: Cert_5_5_08_SplitRoutersLostLeader.py (example: [travis master build](https://travis-ci.org/abtink/openthread/jobs/318196314))
- FAIL: Cert_9_2_12_Announce.py

LuDuda, 2018-01-08:
Adding logs to already reported test.

FAIL: Cert_9_2_12_Announce.py - [travis_9_2_12_announce.txt](https://github.com/openthread/openthread/files/1613163/travis_9_2_12_announce.txt)

librasungirl, 2018-01-09:
FAIL:  Cert_5_1_09_REEDAttachConnectivity
https://travis-ci.org/openthread/openthread/jobs/326701681

librasungirl, 2018-01-09:
FAIL: Cert_5_1_08_RouterAttachConnectivity
https://travis-ci.org/openthread/openthread/jobs/326715975

librasungirl, 2018-01-13:
FAIL：Cert_5_2_5_AddressQuery
https://travis-ci.org/openthread/openthread/jobs/328382463
@hjian2017 , maybe you want to have a look as well as it is you who last updated it #2173

librasungirl, 2018-01-13:
Another failing run of Cert_9_2_12_Announce in @jwhui 's #2472 , with the same error as the log @LuDuda attached above.

> FAIL: test (__main__.Cert_9_2_12_Announce)
> ----------------------------------------------------------------------
> Traceback (most recent call last):
>   File "../../../../tests/scripts/thread-cert/Cert_9_2_12_Announce.py", line 124, in test
>     self.assertEqual(self.nodes[MED].get_state(), 'child')
> AssertionError: 'detached' != 'child'

librasungirl, 2018-02-02:
FAIL: Cert_5_3_05_RoutingLinkQuality when building #2521
[travis_5_3_05_fail_2.txt](https://github.com/openthread/openthread/files/1687973/travis_5_3_05_fail_2.txt)

closed 2018-09-24 by jwhui