openthread/openthread #2444: [mle] multicast address subscription by SEDs

LuDuda, 2017-12-29:
As Thread specification says:

"_Thread devices MAY subscribe to additional realm-local multicast addresses. Rx-off-when-idle Children MUST register such IPv6 multicast addresses with their Parent to receive them via IEEE 802.15.4 indirect transmissions._"

Currently, OpenThread does not allow this operation (registration of multicast address as unicast does not work because of several reasons - e.g. wrong source address selection algorithm operation on SED, mesh forwarder actions on Parent etc..). So the only way to send MPL messages to the SEDs is to use the Realm-Local All Thread Nodes multicast address - which not always is the desirable solution.

I just want to emphasize the lack of this feature. We would be happy to implement this feature, but not in the short time, so please keep this issue open, or if anyone interested to fill the gap, feel free to contribute.

librasungirl, 2018-01-03:
Thanks for raising this. Agree that it is a missing feature. I'd like to implement this.

librasungirl, 2018-01-13:
@LuDuda , #2474 is submitted to provide this feature. I have done some simple validation. Please help to have a review, and maybe further verification if you'd like. Thanks.

LuDuda, 2018-01-13:
@librasungirl that is so awesome! ! Of course i will take a look on your contribution.

PR #2474 (2018-01-29): [netif] support multicast address subscription by SEDs
Support additional link-local/realm-local or higher scope multicast address subscription by SEDs and relative indirect transmission.

To resolve #2444
closed 2018-01-29 by jwhui