openthread/openthread #2445: No Spi Support for EFR32 Platform (feature request)

gemini-dakota, 2017-12-29:
It appears that support for using SPI (as opposed to UART) is not supported for the EFR32 platform. I am able to get UART compiling for using the EFR32 as either a CLI or NCP device. However, when I enable SPI, I receive the following linking error: "undefined reference to `otPlatSpiSlaveEnable'". I briefly looked into this error and openthread appears to be missing the implementation of this function. There is a spi-slave.h, but no spi-slave.c in the efr32 platform. 

Assuming I'm not missing something, I'd like to request the feature addition of SPI support for the EFR32 when used as a NCP (network co processor). 

Otherwise, if this should already be support, let me know as I must be doing something wrong.

jwhui, 2018-01-03:
@gemini-dakota, the EFR32 example drivers do not currently include SPI support.  You are correct, that the platform APIs in`spi-slave.h` need to be implemented.  We welcome any contributions!

gemini-dakota, 2018-01-05:
@jwhui Thanks for the response!

unitrust1, 2018-02-12:
@jwhui you are very correct,the platform APIs in spi-slave.h need to be implemented.

jwhui, 2021-04-07:
Closing stale issue. If this is still of interest to anyone, please open an issue in [openthread/ot-efr32](https://github.com/openthread/ot-efr32).

closed 2021-04-07 by jwhui