openthread/openthread #2470: How to download Thread-Test-Harness1.1-Alpha v1.0-Release_40.0

philixwan, 2018-01-12:
Hello, We want to use the automatic test for openthread. Can you help me where we can download Thread-Test-Harness1.1-Alpha v1.0-Release_40.0?

jwhui, 2018-01-12:
Follow the instructions here: http://graniteriverlabs.com/thread/

philixwan, 2018-01-12:
But I can download just the latest version from here, Is this version used as Harness Automation Tool?

jwhui, 2018-01-12:
You should use the latest version available at the link above.

librasungirl, 2018-01-12:
There are many updates in TH since Alpha v1.0-Release_40.0. It may not have much sense to run with this out-of-date version.
We have realized that the automation tool may not work well with the latest TH R1.2.0 and I believe @FiveDimensions is working on this. Please help to update when there is any progress @FiveDimensions, thanks!
Before the automation tool is ready, TH itself provides another AUTO_DUT way to accelerate the regression. Please refer to Test Harness Auto DUT mode (https://openthread.io/certification/automation_setup) or TH Quick-Start-Guide for details.
Hope it helps.

philixwan, 2018-01-12:
Thank you very much.

closed 2018-01-16 by jwhui