openthread/openthread #2498: Disable EFR32 UART hardware flow control

al1img, 2018-01-22:
In PR https://github.com/openthread/openthread/pull/2401 hardware flow control was enabled for UART because it is recommended by Spinel specification. But pyspinel and borderrouter don't configure HW flow control by default, neither other platforms has flow control enabled by default. So, NCP setup doesn't work out of the box and required additional configuration for EFR32.

The proposal is to disable HW flow control for EFR32 for now. To follow Spinel specification recommendation it should be enabled for all platforms once pyspinel and borderrouter set HW flow control by default. Or any other complex solution. Also flow control should not be enabled for CLI application by default.

jwhui, 2018-01-23:
@tomas-c recommended and tested the use of HW flow control in #2401.  @tomas-c, can you comment on the proposal above?

tomas-c, 2018-01-23:
Enabling flow control solved most of my NOMEM issues in long-running setups. The spinel specification not only recommends but requires flow control.

I quickly checked NRF52840 code and it seems it enables flow control too (UART_HWFC  is set to NRF_UART_HWFC_ENABLED). Unless I'm missing something?

On the other hand, I agree that it is unnecessary for CLI version.

How about instead of setting uartdrvFlowControlHwUart directly in uart.c, we use HAL_SERIAL_APP_FLOW_CONTROL from hal-config.h file? And maybe note in the platform README.md that we highly recommend flow control in NCP mode?

al1img, 2018-01-24:
Probably the issue name is confusing. I don't mean to disable flow control completely but temporary because the main issue I see it is inconsistency between different components of openthread:
- not all platforms enable hw flow control by default (only efr32 and nrf);
- border router doesn't enable hw flow control for wpantund;
- pyspinel doesn't enable hw flow control;
- openthread documentation doesn't mention that flow control should be enabled;
- a user doesn't expect flow control to be enabled for CLI application.

The proposal was to disable flow control till above inconsistencies are resolved or to resolve these issues step by step.

al1img, 2018-01-25:
I see issue https://github.com/openthread/pyspinel/issues/28 for pyspinel already.

jwhui, 2018-01-25:
@al1img, thanks for clarifying.  I agree that we need to resolve the inconsistencies.  Would you be willing to help out with that?

al1img, 2018-01-26:
Sure I will help. But I would like to agree the plan before make any changes. Also I will rename the issue to better match its topic. The plan is:

1. Extend otPlatUartEnable(void) with two following parameters:
    * uint32_t aBaudRate - UART baud rate;
    * bool aHwFlowControl - enables HW flow control.
CLI: `otPlatUartEnable(115200, false);` NCP: `otPlatUartEnable(115200, true);`
I can implement new otPlatUartEnable function for EFR32 and SAMR21. I don't have other HW to test. So, would be nice to contact to each platform responsible person and ask to implement HW flow control. Or I can put compile warning (or error) that this functionality is not implemented for specific platform.
This is probably the most hard item.

2. Enable HW flow control in pyspinel by default and add possibility to disable it with command line. Related to  openthread/pyspinel#28.

3. In borderrouter configure HW flow control for wpantund in startup script.

4. Update documentation if needed.

If there is no objection I will prepare pull requests.

tomas-c, 2018-01-29:
Regarding point 1: spinel documentation allows software flow too control so perhaps instead of true/false use enum/constants?

jwhui, 2018-01-29:
@al1img, this sounds like a reasonable plan to me.

jwhui, 2018-01-29:
@al1img, had a discussions with some other Nest folks.  Rather than changing `otPlatUartEnable(void)` to accept baud rate and flow control parameters, how about making them platform-specific build-time configurations instead?  The thinking is that OpenThread core on the embedded side does not really need to (and thus should not) be concerned with baud rate and flow control and should be contained within the platform.

al1img, 2018-01-29:
@jwhui, thanks for discussions. In general, UART build-time configuration sounds reasonable. In current implementation it is build-time and platform-specific in some way. What is missing it is make it also application specific in order to have different configuration for NCP and CLI. For this purpose openthread-core-config.h could be extended with application specific config:
```
+#ifdef OPENTHREAD_APP_CORE_CONFIG_FILE
+#include OPENTHREAD_APP_CORE_CONFIG_FILE
+#endif

#ifdef OPENTHREAD_PROJECT_CORE_CONFIG_FILE
#include OPENTHREAD_PROJECT_CORE_CONFIG_FILE
#endif
```
Then in the application config file (openthread-core-ncp-config.h and openthread-core-cli-config.h) UART config could be defined as following:
```
#define OPENTHREAD_CONFIG_UART_FLOW_CONTROL OPENTHREAD_CONFIG_UART_FLOW_CONTROL_HW
#define OPENTHREAD_CONFIG_UART_BAUD_RATE 115200
```

OPENTHREAD_CONFIG_UART_FLOW_CONTROL could take enum values (NONE, HW, SW as suggested by @tomas-c). In case unsupported flow control is selected, build time error could be generated.

What do you think?

jwhui, 2018-01-31:
@al1img, can we just define a common include header that is included by code in [examples/platforms](https://github.com/openthread/openthread/tree/master/examples/platforms)?  I understand the desire for common build options that can be leveraged across platforms.  However, I'm not sure I see why we need to make any updates to OpenThread core.

al1img, 2018-01-31:
@jwhui, sorry I miss your point. The main goal I would like to achieve is different default UART configuration for NCP anc CLI applications. I don't see how it could be done with common include header in examples/platforms only. Adding possibility to configure UART by config header doesn't make much sense compared to current implementation.

jwhui, 2018-01-31:
@al1img, I was thinking the new include in `examples/platforms` would specify defaults and app-specific includes in `examples/apps/cli` and `examples/apps/ncp` could override those defaults.  Of course, code in `examples/platforms` would need to be updated to include these.  Primary concern here is to avoid incorporating mechanisms within OpenThread core that are really specific to platforms or examples.

Does that help?

al1img, 2018-02-01:
Thanks for clarification. The drawback of such solution is creating another configuration file for UART only. There is one place where all configuration done, it is PROJECT_CORE_CONFIG_FILE which overrides default core config and obviously UART configuration should be done there. My proposal was to add possibility to override some settings by specific application. This solution is more generic. It would allow to have different openthread settings for different applications (not only different UART settings but different number of buffers etc.). But I will think about other solution which doesn't change core files.

al1img, 2018-02-12:
Update: currently all platform specific code is compiled to a platform lib and then linked with ncp or cli application. It means that to implement above solution with uart header config two different platform libs are required. Which requires to make changes in build system. Basically it will not be the easy and beauty solution.

jwhui, 2018-02-13:
@al1img what about adding an API to [examples/platforms/platform.h](https://github.com/openthread/openthread/blob/master/examples/platforms/platform.h) that allows for dynamically configuring the UART?

al1img, 2018-02-25:
Technically this is possible: for example to add following function PlatformSetUartConfig(int speed, UartFlowControl flow) which stores UART config in some variables and this config will be applied on otPlatUartEnable call. But IMHO it will look like some kind of workaround.

jwhui, 2018-02-27:
@al1img, I agree that the approach is not ideal.  At the same time, we really want to minimize the interface between OpenThread and the platform layer - as a result, we want to limit the `otPlat*` APIs to represent only the functionality that OpenThread actually uses.  Open to other suggestions if you have them.

nigelg-silabs, 2019-08-08:
I'm not sure that this is an EFR32 issue per se. EFR32 enables flow control by default (except on the BRD4166 which has it disabled due to a feature of this demo board)

jwhui, 2019-08-15:
Closing stale issue.

closed 2019-08-15 by jwhui