openthread/openthread #2501: [DTLS] Cannot successfully join to Thread network thru own written code

VictorAtPL, 2018-01-23:
Hi,

I have two CC2538 modules. I have managed to successfully join to commissioner when both devices are configured by CLI. It weren't 100% rate success joins, but the trend was that I could join and ping the another device.

I started to write custom main.c files, first is for commissioner and second is for joiner.

Unfortunately when I start devices, the network is being formed, but the joiner couldn't join.

Even if I just form network with commissioner code and join to it by CLI on joiner - it doesn't work too.

The problem seems to be with MbedTls and/or my code. I see it hangs for some time on 

> generating new ecjpake parameters

and in that time it doens't send ACKs to other device (frame tx failed; failed to send ipv6; Dropping (reassembly queue) IPv6 UDP msg). Does CC2538 example have implemented SHA software generating? Because I suppose some kind of keys are being generated while establishing encrypted connection between devices?

I attach pcap file (wireshark-compatible) where all packets are sniffed. Also I attach pcap file from successful joining when I was doing it by CLI on both devices. Also I give you debug logs and main.c files, so please be so kind to check it (especially otJoinerCallback - is it implemented well?).
[commissioner_debug.log](https://github.com/openthread/openthread/files/1654359/commissioner_debug.log)
[joiner_debug.log](https://github.com/openthread/openthread/files/1654360/joiner_debug.log)
[main.c files](https://gist.github.com/VictorAtPL/7595ef2a2cd929696da22084aede39ac)
[from start, thru succesful join to 5 pings](http://piotrpodbielski.pl/openthread/thread_wystartowanie-po-dolaczenie-i-5-ping.pcapng)
[from start to unsuccessful join](http://piotrpodbielski.pl/openthread/packets.pcapng)

I would appreciate any help,
Piotr

VictorAtPL, 2018-01-25:
Any thoughts guys? Or suggests of ways how to debug it?

Need to make it work, because it's part of my thesis paper.

jwhui, 2018-01-25:
@VictorAtPL, I did a quick review of your code and did not notice anything obvious.  I also tested your code with posix platform and it seemed to work fine.  I suspect that this is an issue specific to the CC2538 platform.

The Joiner Finalize message is used to trigger application and/or vendor-specific provisioning.  See Thread 1.1.1 Section 8.4.5 for more details.

The "hangs" you see during mbedtls crypto operations are expected.  The CC2538 example drivers do not implement platform-specific acceleration of elliptic curve crypto operations.

In your "update", you indicated that the same issue occurs with a CLI-based joiner built clean from GitHub master branch?

jwhui, 2018-01-31:
@VictorAtPL, I just tried your main.c files on top of the latest master with two CLI nodes (https://github.com/openthread/openthread/commit/9190aa21c4b3983c05aad6be8fd5d62942b8cbd8) and everything worked as expected.

I built the images using: `COMMISSIONER=1 JOINER=1 make -f examples/Makefile-cc2538`

jwhui, 2018-02-16:
Closing this issue.  Please reopen if you still have this issue.

closed 2018-02-16 by jwhui