openthread/openthread #2507: otThreadGetDeviceRole returns OT_DEVICE_ROLE_CHILD long after the router has timed it out

GiedriusM, 2018-01-24:
Hi, my team has encountered a peculiar bug which is reproduced in non-standard way, so I expect some grain of salt when looking at it. There's a setup with NCP (border) router and few (2-4) devices, all of which are custom platform. Some devices are connected to a debugger and may be paused for indefinite amount of time. If the pause is long enough that the router times-out the device, it does not reconnect once resumed. Capturing the packets show that the device only sends data requests (and receives ACKs to them). Calling otThreadGetDeviceRole always return OT_DEVICE_ROLE_CHILD, instead of OT_DEVICE_ROLE_DETACHED.
Has anyone noticed similar behavior? Could it be enough that the ACKs to data requests are enough to keep the child from switching to detached state? I have not yet looked deep enough to determine what's the exact problem here, it may as well be platform or debug related, but any suggestions on where to look first in openthread stack are welcome.

jwhui, 2018-01-24:
Yes, this is expected behavior for Thread.

Thread 4.7.5 states:
> Rx-off-when-idle Children SHOULD also periodically (at time intervals indicated by the application) resynchronize by sending a Child Update Request to their Parent to avoid cases when the MAC Data Request is being acknowledged erroneously by a receiver that is not a valid Parent.

OpenThread includes an optional build feature that implements the above.  You can enable the build feature by passing the `--enable-child-supervision` flag as part of configure.  Some more pointers regarding this feature:
- APIs in [include/openthread/child_supervision.h](https://github.com/openthread/openthread/blob/master/include/openthread/child_supervision.h)
- Implementation in [src/core/utils/child_supervision.cpp](https://github.com/openthread/openthread/blob/master/src/core/utils/child_supervision.cpp)

jwhui, 2018-01-31:
@GiedriusM, do you have anything else to add to this issue?

GiedriusM, 2018-02-01:
Thank you, @jwhui! It completely answers my question.

closed 2018-02-01 by jwhui