openthread/openthread #2530: [efr32] otPlatRadioTxDone not being called

tomas-c, 2018-02-02:
In some of the larger networks wpantund started printing "STATUS_NOMEM" errors. After this happens, the devices become leaders and form one device partitions. Checking the status of application buffers shows they are all being used for transmission and are not being released.

I made the modifications to radio.c to debug the problems. I have attached the diff at the bottom (the line numbers might not match exactly as I have more log messages, not shown here).

What I noticed is that at some point otPlatRadioTxDone is never called. Wpantund gets stuck printing
`NCP => [CRIT]-PLAT----: TX taking > 1000ms` (the log message I added) every second.

OpenThread version: 68a605f22e579ae45ab1d8221faa2d45e8668e05
Wpantund version: bcee1206c3e73b82f5e5cb16e624bd83b0114ffa

```C++
@@ -83,6 +83,9 @@ static otRadioFrame   sTransmitFrame;
 static uint8_t        sTransmitPsdu[IEEE802154_MAX_LENGTH];
 static otError        sTransmitError;
 
+static bool           sTransmitWaiting    = false;
+static uint32_t       sTransmitStart      = 0;
+
 typedef struct        srcMatchEntry
 {
     uint16_t checksum;
@@ -358,6 +361,10 @@ otError otPlatRadioTransmit(otInstance *aInstance, otRadioFrame *aFrame)
     status = RAIL_StartCcaCsmaTx(sRailHandle, aFrame->mChannel, txOptions, &csmaConfig, NULL);
     assert(status == RAIL_STATUS_NO_ERROR);
 
+    assert(sTransmitWaiting == false);
+    sTransmitStart = otPlatAlarmMilliGetNow();
+    sTransmitWaiting = true;
+
     otPlatRadioTxStarted(aInstance, aFrame);
 
 exit:
@@ -818,6 +825,14 @@ otError otPlatRadioEnergyScan(otInstance *aInstance, uint8_t aScanChannel, uint1
 
 void efr32RadioProcess(otInstance *aInstance)
 {
+    if(sTransmitWaiting == true) {
+        int msWaiting = otPlatAlarmMilliGetNow() - sTransmitStart;
+        if(msWaiting >= 1000) {
+        	otLogCritPlat(sInstance, "TX taking > %dms", msWaiting);
+        	sTransmitStart = otPlatAlarmMilliGetNow();
+        }
+	}
+
     if (sState == OT_RADIO_STATE_TRANSMIT && sTransmitBusy == false)
     {
         if (sTransmitError != OT_ERROR_NONE)
@@ -838,6 +853,8 @@ void efr32RadioProcess(otInstance *aInstance)
         else
 #endif
         {
+            assert(sTransmitWaiting == true);
+            sTransmitWaiting = false;
             otPlatRadioTxDone(aInstance, &sTransmitFrame, NULL, sTransmitError);
         }
     }
```

jwhui, 2018-02-05:
@tomas-c, thanks for raising this issue.

The only things that come to mind are:
1. OT are not properly handling one or more RAIL events that indicate the end of transmission.
2. OT is making calls to the RAIL API that causes the transmission to abort without RAIL sending a `RAIL_EVENT_TX_ABORTED`  event.
3. The RAIL library has a bug or is behaving in an unexpected way.

tomas-c, 2018-02-05:
Here's what I currently know:

**Regarding option 1)**

From Rail 2.1 rail_types.h (Rail 2.0 doesn't define RAIL_EVENTS_TX_COMPLETION)

```
/**
 * This mask indicates all of the events that determine the end of a transmitted
 * packet. After a packet has begun to be transmitted, which is indicated by
 * a \ref RAIL_STATUS_NO_ERROR return value from one of the transmit functions,
 * exactly one of the following events will occur. When one of these events
 * occurs a state transition will take place based on the parameter
 * passed to RAIL_SetTxTransitions(). The RAIL_StateTransitions_t::success
 * transition will be followed only if the \ref RAIL_EVENT_TX_PACKET_SENT
 * event occurs. Any of the other events will trigger the
 * RAIL_StateTransitions_t::error transition.
 */
#define RAIL_EVENTS_TX_COMPLETION (RAIL_EVENT_TX_PACKET_SENT \
                                   | RAIL_EVENT_TX_ABORTED   \
                                   | RAIL_EVENT_TX_BLOCKED   \
                                   | RAIL_EVENT_TX_UNDERFLOW \
                                   | RAIL_EVENT_TX_CHANNEL_BUSY)
```

We currently handle all of these and on every one of them set `sTransmitBusy = false;`. The only exception is when we receive RAIL_EVENT_TX_PACKET_SENT where we check for `(sTransmitFrame.mPsdu[0] & IEEE802154_ACK_REQUEST) == 0`. 

I checked and in every single case, the packet that gets stuck in transmission does not require an ACK. (It's a bit strange actually - never saw a packet that requires an ACK to get stuck).

**Regarding option 2)**

When the issue occurs, I print RAIL_GetRadioState(sRailHandle) every second and it's always equal to RAIL_RF_STATE_TX. Not completely sure how rail works in this regard, but I'd expect the radio to switch to reception or idle mode upon failed TX.

If it's actually a bug in Rail then it might be difficult to relay it to SiLabs as it's quite difficult to reproduce and we don't have Rail source code.

tomas-c, 2018-02-07:
After more debugging I noticed that, just before the problem happens, RAIL_EVENT_RX_FIFO_OVERFLOW is triggered.

My first guess was that if a frame requires an ACK and RAIL_EVENT_RX_FIFO_OVERFLOW is triggered then, depending on how Rail is implemented, it's possible RAIL_EVENT_RX_ACK_TIMEOUT is never called and efr32RadioProcess thinks that transmission never finished. The only problem with this theory is that, in every test, the frame that gets stuck in transmission does not require an ACK.

Nevertheless, I removed `otEXPECT(sState == OT_RADIO_STATE_RECEIVE);` from processNextRxPacket function and added `otEXPECT(status != RAIL_STATUS_INVALID_STATE);` just after RAIL_GetRxPacketDetails() call. This made sure that the RX FIFO is always drained, even if the radio is struck in transmission.

@jwhui Any guesses why this worked?

I'll keep my devices running for a few days and then will make a PR if there aren't any issues.

jwhui, 2018-02-07:
@tomas-c, thanks for investigating this issue!  Given that calls to `RAIL_StartCcaCsmaTx()` does not return an error, my guess is that RAIL actually delays the transmission until after the RX FIFO exits the overflow state.  As a result, with your fix, we will flush the RX FIFO properly and allow RAIL to perform the transmission.

jwhui, 2018-02-08:
Resolved by #2549.

closed 2018-02-08 by jwhui