openthread/openthread #2537: [question] otLinkSetExtendedAddress and IEEE EUI-64

GiedriusM, 2018-02-03:
Hi,

Thread specification explicitly prohibits the use of a factory assigned IEEE EUI-64 (Table 3-2). The only exception is out-of-band commissioning when the extended address MAY be set to IEEE EUI-64, however given the nature of "out-of-band" it is not going to be used anyway.

So more out of curiosity, rather than technical reason, I have few questions:
1) If IEEE EUI-64 is forced as an extended address during commissioning or normal operation, what kind of issues can it cause? Can it compromise network security?
2) What if IEEE EUI-64 is exposed to public by other means, but is not used directly as a an extended address?
3) Given the fact, that OpenThread takes care of generating random addresses, is there any use case where otLinkSetExtendedAddress should be used?

jwhui, 2018-02-05:
The IEEE EUI-64 identifier is an administered identifier typically configured in the factory.  While factory-assigned identifiers are typically assumed to be globally unique, there are a number of instances where that assumption was violated (flaws in manufacturing, counterfeit devices, etc.).

In the context of Thread, devices using the same IEEE Extended Address in the same network leads to the following issues:
1. IEEE 802.15.4 MAC security uses Extended Addresses in the Nonce and relies on having unique Extended Addresses so that a give Nonce value is used at most once for a given AES-128 key.
2. MLE uses Extended Addresses to uniquely identify neighboring devices.  If two devices attempt to attach with the same Extended Address, messages will be dropped due to conflicting frame counters.

For these reasons, the broader industry (IETF, Thread Group, etc.) are migrating away from using factory-assigned unique identifiers towards stable but randomly-chosen identifiers.  From past industry experience, the likelihood of two devices randomly selecting the same identifier is much less likely than the occurrence of duplicates in factory-assigned identifiers.

Regarding the use case for `otLinkSetExtendedAddress()`, note that IPv6 over IEEE 802.15.4 communication may occur before Thread protocols are enabled via `otThreadSetEnabled()`.  Nest uses this feature to support some legacy protocols.

GiedriusM, 2018-02-05:
Thanks, @jwhui, once again it answers my questions! "Out-of-band" meaning same "band", but different protocol did not come into my mind :)

closed 2018-02-06 by jwhui