openthread/openthread #2538: [efr32] otPlatRadioTxDone ACK frame is always NULL

tomas-c, 2018-02-04:
I see that when reporting end of transmission in efr32 platform driver, the ACK frame is always NULL. Other platforms pass the ACK frames they receive to otPlatRadioTxDone. 

Can not doing so cause issues?

jwhui, 2018-02-05:
@tomas-c, thanks for raising this issue.  Passing the ACK frame up is required for OpenThread to process the Frame Pending bit in an IEEE 802.15.4 ACK frame.  Note that this mechanism is only required when operating as a Sleepy End Device (SED).  In any case, the ACK frame should be passed up to OpenThread for completeness.  Would you like to submit a PR to address this issue?

tomas-c, 2018-02-05:
Will do if I ever get #2530 solved - that's a bit higher on the priority list.

jwhui, 2018-03-14:
@tomas-c, I just submitted #2617, which should resolve this issue.  Please review if you get a chance.

PR #2617 (2018-03-21): [efr32] pass up received ACK frame
Resolves #2538.
closed 2018-03-21 by jwhui