openthread/openthread #2548: [SAMR21] error: this use of "defined" may not be portable

TobiasReichert, 2018-02-07:
Hey guys,
I'd like to build openthread on _SAM R21 Xplained Pro_ .
I have made the first two steps from the build example without any problems.
But I don't know what to do on the third or fourth step.
There is no _conf_board.h_ file or _CONF_IEEE_ADDRESS_ macro.
And in the fourth step you should change a _user_board.h_ which I also couldn't find.

Ignoring these steps and building like it is descried int the fifth step will bring me an error like:
'
[...]
/home/tobias/Dokumente/openthread/examples/../third_party/microchip/asf/sam0/drivers/sercom/usart/usart.h:287:1: error: this use of "defined" may not be portable [-Werror=expansion-to-defined]
(SAMR30) || (SAMHA1) || defined(__DOXYGEN__)
^~~~~~~~~~~~~~
/home/tobias/Dokumente/openthread/examples/../third_party/microchip/asf/sam0/drivers/sercom/usart/usart.h:287:1: error: this use of "defined" may not be portable [-Werror=expansion-to-defined]
/home/tobias/Dokumente/openthread/examples/../third_party/microchip/asf/sam0/drivers/sercom/usart/usart.h:287:1: error: this use of "defined" may not be portable [-Werror=expansion-to-defined]
/home/tobias/Dokumente/openthread/examples/../third_party/microchip/asf/sam0/drivers/sercom/usart/usart.h:287:1: error: this use of "defined" may not be portable [-Werror=expansion-to-defined]
/home/tobias/Dokumente/openthread/examples/../third_party/microchip/asf/sam0/drivers/sercom/usart/usart.h:287:1: error: this use of "defined" may not be portable [-Werror=expansion-to-defined]
/home/tobias/Dokumente/openthread/examples/../third_party/microchip/asf/sam0/drivers/sercom/usart/usart.h:287:1: error: this use of "defined" may not be portable [-Werror=expansion-to-defined]
(SAMR30) || (SAMHA1) || defined(__DOXYGEN__)
^~~~~~~~~~~~~~~
/home/tobias/Dokumente/openthread/examples/../third_party/microchip/asf/sam0/drivers/sercom/usart/usart.h:287:1: error: this use of "defined" may not be portable [-Werror=expansion-to-defined]
/home/tobias/Dokumente/openthread/examples/../third_party/microchip/asf/sam0/drivers/sercom/usart/usart.h:287:1: error: this use of "defined" may not be portable [-Werror=expansion-to-defined]
[...]
'

What am I doing wrong?
Hope you can help =)

OS: Ubuntu 16.04.3 LTS
arm-none-eabi-gcc version: 7.2.1 20170904

al1img, 2018-02-07:
Hi @TobiasReichert , about steps to build: the third step is outdated, currently IEEE extended address is read correctly form user row section for SAMR21 Xplained Pro. Step 4 is for boards different from Xplained Pro. So, just skip steps 3 and 4. In case Xplained Pro board it should be compiled without changes. 

About error you see: the latest gcc has more strict rules. There is no such error with gcc 6. You can switch to gcc 6 or try to add `-Wno-expansion-to-defined` to `libopenthread_samr21_a_CPPFLAGS` variable in [examples/platforms/samr21/Makefile.am](https://github.com/openthread/openthread/blob/master/examples/platforms/samr21/Makefile.am). Do not forget to call `./bootstrap` after changes in the make file.

jwhui, 2018-02-07:
@al1img, thanks for the clarifications!  Would you be willing to update [examples/platforms/samr21/README.md](https://github.com/openthread/openthread/blob/master/examples/platforms/samr21/README.md) with this info?

al1img, 2018-02-07:
Sure, I will update. Also will check the compiling issue with gcc 7 and add flag if it works.

jwhui, 2018-02-09:
Documentation updated in #2552.

closed 2018-02-09 by jwhui

TobiasReichert, 2018-02-12:
Okay fixed =)
Thank you very much @al1img @jwhui for your friendly help