openthread/openthread #2568: [kw41z] OPENTHREAD_ENABLE_CLI_LOGGING no longer supported

rdewaele, 2018-02-18:
The example platforms _kw41z_ and _efr32_ include a header `<cli/cli-uart.h>` and a call to a function `otCliLog` that are enabled when the macro `OPENTHREAD_ENABLE_CLI_LOGGING` is defined. 

It seems that the header nor the function no longer exist, per https://github.com/openthread/openthread/commit/da357701406a38a94478b7f8c6a90f09cded0166

However, `src/cli/cli_uart.cpp` still supports `extern "C"`, so should this header be recreated and where?

rdewaele, 2018-02-18:
Adding a C header with prototypes corresponding to cli_uart.cpp and changing `otCliLog ` to `otCliPlatLogv` fixed the issue for me on _kw41z_.

As I can not be sure about the intent when the header file was removed, I would like some advice as to how this issue should be resolved correctly. I'll gladly make a PR upon feedback on this issue.

jwhui, 2018-02-20:
@rdewaele, thanks for reporting this issue.  The macro `OPENTHREAD_ENABLE_CLI_LOGGING ` and function `otCliLog` is not longer used and these other calls you point out should be removed.  The examples [examples/platforms/efr32/logging.c](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/logging.c) and [examples/platforms/kw41z/logging.c](https://github.com/openthread/openthread/blob/master/examples/platforms/kw41z/logging.c) should be replaced with [examples/platforms/cc2538/logging.c](https://github.com/openthread/openthread/blob/master/examples/platforms/cc2538/logging.c).  Feel free to submit a PR that cleans this up.  Thanks!

PR #2605 (2018-03-13): [logging] remove deprecated use of OPENTHREAD_ENABLE_CLI_LOGGING
Resolves #2568.
closed 2018-03-13 by jwhui