openthread/openthread #2571: Unknown Neighbor during Join

DuaneEllis-TI, 2018-02-20:
This problem has existed on and off for quite some time now - it's also a very temperamental problem - meaning - it goes away, then comes back on its own.

Short version: During commissioning, the joining device gives up with an "Unknown Neighbor" message
the work around seems to be (a) reconfigure the network (new pan id, new passwords) and it seems to fix it self. or wait a little bit and it fixes it self.

Attached are two logs - one is called the NCP.TXT - it represents the debug log from the NCP
The second is joiner.TXT - it represents an MTP device 

What I think is happening is this:  

the precursor - There is or was an earlier failed JOIN operation - for some reason the joiner to be entered into the neighbor table as a valid entry.

The setup - "factory reset" the joining device - now try to rejoin that same device.

In my mind, if and when a device sends a Discovery Request - that should automatically clean up the neighbor table - removing that device from the nieghbor table

====
How did I get this? I enabled the DEBUG UART, and DISABLED the MBED logging ( that generates too much noise)
[ncp.txt](https://github.com/openthread/openthread/files/1742030/ncp.txt)
[joiner.txt](https://github.com/openthread/openthread/files/1742031/joiner.txt)

DuaneEllis-TI, 2018-02-20:
See: https://github.com/openthread/openthread/issues/2299

jwhui, 2018-02-20:
@DuaneEllis-TI, thanks for digging into this issue.

Looking at the ncp.txt log file, it seems to start off with one link-local IPv6 address:
```
110.303 | [INFO]-MAC-----: Sent IPv6 UDP msg, len:94, chksum:9fbf, to:4fe3a7e6a12761db, sec:no, prio:high
110.312 | [INFO]-MAC-----: src: fe80:0:0:0:9878:db1c:96b0:79d5
110.318 | [INFO]-MAC-----: dst: fe80:0:0:0:4de3:a7e6:a127:61db
```

But then switches to using a different one later on:
```
116.081 | [INFO]-MAC-----: Sent IPv6 UDP msg, len:92, chksum:18da, to:4fe3a7e6a12761db, sec:no, prio:high
116.090 | [INFO]-MAC-----: src: fe80:0:0:0:c742:9fc9:f6fe:7264
116.096 | [INFO]-MAC-----: dst: fe80:0:0:0:4de3:a7e6:a127:61db
```

The `MeshForwarder` component [selects the MAC Source Address](https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder.cpp#L422) by first doing a match of the IPv6 Source IID with the Extended Address and, if they do not match, uses the short 16-bit address for the MAC Source Address.

The `JoinerRouter` component  [does not specify an IPv6 source address](https://github.com/openthread/openthread/blob/master/src/core/meshcop/joiner_router.cpp#L285), relying on [default IPv6 source address selection](https://github.com/openthread/openthread/blob/master/src/core/net/ip6.cpp#L1000).  However, it is not yet clear to me why it starts selecting a different source address.

jwhui, 2018-02-20:
@DuaneEllis-TI, is it possible that the linux host is assigning `fe80:0:0:0:c742:9fc9:f6fe:7264` to the Thread interface, which then gets propagated down into the NCP?

DuaneEllis-TI, 2018-02-20:
Attached is the output of "ifconfig" and various other things some of which I had to type by hand...
with various IP addresses  of other parts of the system

[ifconfig-bbb.txt](https://github.com/openthread/openthread/files/1742224/ifconfig-bbb.txt)

DuaneEllis-TI, 2018-02-20:
Oops did not intend to close...

DuaneEllis-TI, 2018-02-20:
Hmm - I formed a new network and might be misleading you.
the NCP log shows the network name as
>> 110.282 | | 22 22 03 05 46 65 62 32 | 30 08 0F 00 00 00 20 04 | ""..Feb20..... .

Which is 'as it was'  when the problem first introduced it self again.

You'll notice the network name has changed from Feb20 - to Feb20_v2
sorry for the confusion.

[ifconfig-v2.txt](https://github.com/openthread/openthread/files/1742305/ifconfig-v2.txt)

Let me back up to the top and and reproduce again...

DuaneEllis-TI, 2018-02-20:
Setting expectations - this will take a day or so - before I can get back to this.

DuaneEllis-TI, 2018-02-21:
Q: If there are some places i could / should instrument with some more logging - (ie: Litter the some section or area with lots of print statements) - please point me at them. 

I do think this is really about a corner/error-recovery case - 

ie: Something goes wrong in the commissioning process.

The scenario I'm thinking is this: 

the Human (our victim) is lost - and tries to reset things to start all over again.
Sort of like trying to do a 'factory reset' - but - has only factory reset one or two things
Perhaps they did not know how to factory reset everything - or they thought they did...

jwhui, 2018-02-21:
@DuaneEllis-TI, it might be useful to log when unicast addresses are added in `Netif::AddUnicastAddress()`.  I suspect that there are multiple link-local IPv6 addresses, causing the default source address selection to use the wrong one.

DuaneEllis-TI, 2018-02-21:
Updated logs - added logs in various other functions also - you should recognize the names, I am also printing the address that is being passed to the function where used/possible.

I have not reviewed these yet.

[joiner-feb21.txt](https://github.com/openthread/openthread/files/1746126/joiner-feb21.txt)
[bbb-feb21.txt](https://github.com/openthread/openthread/files/1746127/bbb-feb21.txt)
[ncp-feb21.txt](https://github.com/openthread/openthread/files/1746128/ncp-feb21.txt)

DuaneEllis-TI, 2018-02-22:
Hmm - the joiner.txt - where it states:

>> 163.199 | [INFO]-MESH-CP-: Sent relay rx
>> 163.929 | [INFO]-MESH-CP-: Received relay transmit
>> 163.931 | [INFO]-MAC-----: short @444
>> 163.934 | [DEBG]-MAC-----: Request to start operation "TransmitData"
>> 163.940 | [DEBG]-MAC-----: Starting operation "TransmitData"

the "short@444" - occurs in the function   MeshForwarder::GetMacSourceAddress()

The current code is here: https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder.cpp#L430

Our internal branch is a little behind (we are in the process of merging to master now)

DuaneEllis-TI, 2018-02-22:
The host - well that's WPANTUND - straight out of the box - no change, so this change would need to occur in wpantund.

DuaneEllis-TI, 2018-02-22:
Remember here - this is occurring during a relay - from the far away Android Device.

In this case, the leader/commissioner is the NCP - 

I don't know enough of the spec to question another possible corner case, specifically: Where the Joiner is multiple mesh/router hops away...  from the NCP/Border Router.

jwhui, 2018-02-22:
The communication between the Joiner Router and Joiner is always done via link-local addresses - doesn't matter how many hops away the Border Router is from the Joiner.

wpantund does not add its own link-local addresses - it simply takes them from either the NCP or linux network interface.  The NCP log clearly indicates that the link-local address is "external" (i.e. coming from the host).  Ideally, the linux host would not add link-local addresses to the wpan0 interface in the first place.  However, as a protective measure, we can have OpenThread reject adding external link-local addresses and somehow propagate that error back to wpantund.

DuaneEllis-TI, 2018-02-22:
So putting it another way...

The NCP ends up with multiple IPv6 addresses and chooses the wrong one here.

Once the joining operation starts ... (ie: Discovery Request/Response) why can't the meshcop code just "lock on-to" the address used during the discovery process?

jwhui, 2018-02-22:
I just submitted #2579.

There's really only one link-local IPv6 address that the device should use.  Thread assumes that link-local addresses are derived from the IEEE 802.15.4 Extended Address.

abtink, 2018-02-22:
We can consider filtering the user/interface added link-local IPv6 addresses on `wpantund` (i.e., not syncing a link-local address from user/interface on NCP). 

I wonder if there is a way to configure linux host to not add its own link-local IPv6 address. 
Maybe disabling autoconf  `echo 0 > /proc/sys/net/ipv6/conf/wpan0/autoconf`

abtink, 2018-02-23:
Just submitted https://github.com/openthread/wpantund/pull/278 on wpantund.

DuaneEllis-TI, 2018-02-27:
As an FYI - the way to reproduce the original issue easily seems to be 

1) using gui form the network.
2) Using the android app commission a device.

3) Important - power cycle, the entire border router (NCP + linux-host) 
I am not sure if is always limited to just the NCP

4) Try to commission another device.

jwhui, 2018-06-06:
Closing due to inactivity.

closed 2018-06-06 by jwhui