openthread/openthread #2585: Copying Headers in OpenThread

samkumar, 2018-02-24:
I noticed that when the Ip6 layer processes a received packet, it first copies the header out of the message to the stack, and then processes the header. Similarly, when sending a packet, it first constructs the header on the stack, and then copies it into the Message. Other layer of the protocol stack (e.g., UDP) work in the same way.

However, if OPENTHREAD_CONFIG_MESSAGE_BUFFER_SIZE is tuned correctly, then all of the headers are likely to be in the first buffer in the Message. The headers won't be split across multiple Message buffers. Therefore, it would be more efficient to just process headers by reading the memory in the Message buffer itself, instead of first copying it to a stack buffer.

This could be accomplished by adding a public method to the Message class:

```
uint8_t* GetPointer(uint16_t aOffset) const;
```

And, of course, one would modify the IP6 and UDP layers to just construct headers (for sending packets) and process headers (for receiving packets) directly in Messages. This would be more efficient than copying messages to the stack.

I would be happy to contribute this fix if the developers are receptive to this idea.

jwhui, 2018-02-27:
Thanks for submitting this proposal.

Couple things to consider:
1. There may be multiple IPv6 headers in a given datagram.  Thread uses IPv6-in-IPv6 encapsulation for MPL.  Applications may also choose to use IPv6-in-IPv6 encapsulation.  How do you propose to handle this case?
2. Are you proposing to require that `OPENTHREAD_CONFIG_MESSAGE_BUFFER_SIZE` be large enough to hold an IPv6 + IPv6 HBH Options + IPv6 + UDP header all in the first message buffer?

jwhui, 2018-04-16:
Closing stale issue.  Please reopen if/when you want to push this forward.

closed 2018-04-16 by jwhui