openthread/openthread #2592: PlatformInit() namespacing

robertcragie, 2018-02-27:
PlatformInit() namespacing is a bit generic and has caused an issue in a porting exercise I am doing, Would it be possible to change it to otPlatInit() in line with the other platform files? The same would apply to PlatformProcessDrivers().

jwhui, 2018-02-27:
@robertcragie, thanks for raising this issue.

We want to reserve the `otPlat*` prefix for platform functions that the OpenThread library actually uses.  Things like `PlatformInit()` and `PlatformProcessDrivers()` are specific to the particular examples that are using OpenThread.  Would something like `otExampleInit()` and `otExampleProcessDrivers()` help?

turon, 2018-02-27:
I'd like to throw the proposed namespace of `otRuntime*` into the hat.  So `PlatformInit()` and `PlatformProcessDrivers ()` would become `otRuntimeInit()` and `otRuntimeProcessDrivers()`.

jwhui, 2018-02-27:
@turon, not sure I see "runtime" as a good fit.  A "runtime" typically implements an execution model, or portions thereof.  In this case, `PlatformInit()` and `PlatformProcessDrivers()` have nothing to do with the execution model.

robertcragie, 2018-02-28:
@jwhui I see your point as these are called by the example apps' `main()`, not the OpenThread library and are indeed specific to the way the examples are structured. The `otExample` prefix works for me but I would suggest making that distinction clear in the porting guide, i.e. separating what is actually platform abstraction from specific example implementation API.

PR #2927 (2018-08-03): [examples] change example platform namespacing
Resolves #2592.
closed 2018-08-03 by jwhui