openthread/openthread #2595: Invalid Border Agent Locator in leader network data (unable to join)

mwilniewiec, 2018-02-28:
I found that leader + commissioner can eventually go to an invalid state where Commissioner is sending keep alives and it is accepted by the leader but nobody can join.

The problem shows up when a border agent sent a leader petition in a child state (not leader, nor router). In that case a child RLOC is saved in network data Border Agent Locator.
When this thread device changes it's state to "router" its RLOC is changed but no-one updates this field in leader network data.

From this point a border agent sends LEAD_KA to the leader with commissioner session ID and gets valid response (Accept) but the border agent locator in the network data points to the device that doesn't exist.

I reproduced the issue using 3 ot_cli_ftd.
1. Run two clis and make sure that one is a leader and one is a child.
2. On the child cli make commissioner start
3. Make sure that it is still a child and verify netdatashow - this device RLOC should be found in network data. At this point any device can join as commissioner RLOC is the same as the one in network data.
4. Wait until commissioner changes state to router or force it by "state router"
5. Network data doesn't change and no other device can successfully join - never ...

I couldn't find any related text in Thread docs but I think that leader could check border agent locator when he receives LEAD_KA.

jwhui, 2018-02-28:
@mwilniewiec, thanks for reporting this issue!

I agree that the Thread specification does not address how to handle RLOC changes.  The Thread Group is tracking this issue in [SPEC-490](https://threadgroup.atlassian.net/browse/SPEC-490), but that has gone stale.

I do agree that, at a minimum, that the Leader should update the Border Agent Locator TLV when noticing a change via the Keep Alive messages.  I've submitted #2596 to address this issue.

However, even with #2596, changes in the RLOC will go unnoticed until the next Keep Alive message is sent.  I see two options:
1. Send a Keep Alive message immediately after a role change.
2. Do not allow role changes while the device is a Commissioner.

Thoughts?

Nopik, 2018-03-03:
I don't think that the 2) approach (not allowing changes) will be a good solution. I would just go with #2596 and allowing mesh to heal after short period of time. Then, as optimization, reduce this time to minimum, hopefully to zero.

PR #2596 (2018-03-07): [meshcop] update Border Agent Locator TLV on keep alives
Resolves #2595.
closed 2018-03-07 by jwhui