openthread/openthread #2640: Traffic hang issue

zhaxia, 2018-03-27:
Scenario:
Four NORDIC dev boards with ot-cli-ftd image. The image is based on the commit: 4e92a73. The topology is shown in the following figure. A is leader, B, C and D are routers.
         A
          |
         B
       /    \
      C     D

Issue description:
Both the Node C and D are sending traffic to A at the same time. A moment later, the node A can't receive any packets from node C and D. After the traffic from C and D is stoped,  all the pings from node C , D and A to B fail.  Several minutes later, node B, C and D become leader.

After the traffic is stopped, I use the command "bufferinfo" to see the buffer information on node B:
> bufferinfo
total: 128
free: 0
6lo send: 64 128
6lo reas: 0 0
ip6: 0 0
mpl: 0 0
mle: 0 0
arp: 0 0
coap: 0 0
coap secure: 0 0
application coap: 0 0
Done

The buffer of the node B is full all the time. After I debug the code, I find that the variable mSendBusy in class MeshForwarder is set to TRUE and its value never changed. Which causes messages are blocked in the queue.
Logs for reference:
[4e92a73_A_rx_log.txt](https://github.com/openthread/openthread/files/1851879/4e92a73_A_rx_log.txt)
[4e92a73_B_log.txt](https://github.com/openthread/openthread/files/1851881/4e92a73_B_log.txt)
[4e92a73_C_tx_log.txt](https://github.com/openthread/openthread/files/1851883/4e92a73_C_tx_log.txt)
[4e92a73_D_tx_log.txt](https://github.com/openthread/openthread/files/1851884/4e92a73_D_tx_log.txt)

However, when I try the commit: 302c999, no traffic hang issue happens.
Logs for reference:
[302c999_A_rx_log.txt](https://github.com/openthread/openthread/files/1851923/302c999_A_rx_log.txt)
[302c999_C_tx_log.txt](https://github.com/openthread/openthread/files/1851925/302c999_C_tx_log.txt)
[302c999_D_tx_log.txt](https://github.com/openthread/openthread/files/1851926/302c999_D_tx_log.txt)

In this scenario, I'm using perf to generate traffic. Perf is a tool I wrote to test the performance of Openthread. It only call Timer, otMessage and otUdp API.

zhaxia, 2018-03-27:
The topology:
[image]

LuDuda, 2018-03-27:
Thank you for reporting this issue. We are looking at it.

LuDuda, 2018-03-29:
We found the root cause in our radio driver. The fix will be available soon. One more time thanks for raising this and very detailed description! 👍

hubertmis, 2018-03-30:
The fix is available here: https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/pull/91
We plan to merge it to OT repository next week (after weekly robustness testing).

PR #2646 (2018-04-04): [nrf52840] update radio driver to version 1.1.0
This PR should fix #2640
closed 2018-04-04 by jwhui