openthread/openthread #2664: Could not send packets any more

yuzhyang, 2018-04-16:
## SUMMARY
One node could not send packets any more after more than 10000 packets sent

## STEPS TO REPRODUCE
Keep sending unicast packets between each possible connection in a 12-node network with 512B payload size, repeat 50 times, interval 500ms
Send multicast packets from each node with 512B payload size, repeat 50 times, interval 500ms
The number of sending packets is more than 10000 and total time is more than 3 hours, then one of the 12 nodes could not send packet any more and lost from the network. Similar to #2640 

Note: If any register info etc. needed from this hung node, please let me know, I am trying to keep the node for more time.

## RESULTS
### Actual: 
1. TX counter would not increase while RX counter keep increasing
2. buffer free is zero
3. lost router info
> counter mac
> TxTotal: 16674
>     TxUnicast: 6142
>     TxBroadcast: 10532
>     TxAckRequested: 6142
>     TxAcked: 6092
>     TxNoAckRequested: 10532
>     TxData: 16674
>     TxDataPoll: 0
>     TxBeacon: 0
>     TxBeaconRequest: 0
>     TxOther: 0
>     TxRetry: 7716
>     TxErrCca: 6384
>     TxErrBusyChannel: 282
> RxTotal: 202516
>     RxUnicast: 6185
>     RxBroadcast: 53654
>     RxData: 59742
>     RxDataPoll: 0
>     RxBeacon: 0
>     RxBeaconRequest: 0
>     RxOther: 0
>     RxAddressFiltered: 0
>     RxDestAddrFiltered: 71681
>     RxDuplicated: 97
>     RxErrNoFrame: 56170
>     RxErrNoUnknownNeighbor: 0
>     RxErrInvalidSrcAddr: 0
>     RxErrSec: 0
>     RxErrFcs: 14826
>     RxErrOther: 0
> 

> counter mac
> TxTotal: 16674
>     TxUnicast: 6142
>     TxBroadcast: 10532
>     TxAckRequested: 6142
>     TxAcked: 6092
>     TxNoAckRequested: 10532
>     TxData: 16674
>     TxDataPoll: 0
>     TxBeacon: 0
>     TxBeaconRequest: 0
>     TxOther: 0
>     TxRetry: 7716
>     TxErrCca: 6384
>     TxErrBusyChannel: 282
> RxTotal: 203616
>     RxUnicast: 6185
>     RxBroadcast: 53904
>     RxData: 59992
>     RxDataPoll: 0
>     RxBeacon: 0
>     RxBeaconRequest: 0
>     RxOther: 0
>     RxAddressFiltered: 0
>     RxDestAddrFiltered: 71819
>     RxDuplicated: 97
>     RxErrNoFrame: 56724
>     RxErrNoUnknownNeighbor: 0
>     RxErrInvalidSrcAddr: 0
>     RxErrSec: 0
>     RxErrFcs: 14984
>     RxErrOther: 0

> bufferinfo
> total: 1024
> free: 0
> 6lo send: 212 1024
> 6lo reas: 0 0
> ip6: 0 0
> mpl: 0 0
> mle: 0 0
> arp: 0 0
> coap: 0 0
> coap secure: 0 0
> application coap: 0 0
> Done
> 

### Expected: 
Long Term Traffic sent without any hung problem

yuzhyang, 2018-04-16:
@LuDuda @hubertmis Please let me know if you want specific register value etc.

LuDuda, 2018-04-16:
Hello @yuzhyang thnks for reporting this issue. Most likely there is some race condition in our radio driver, which incorrectly prevents reporting of ACK timeout or frame being transmitted.

We will of course take a look on that and fix as soon as possible. We will start investigating this at the begginning of the next week due to lack of engineers in next few days (internal activities).

Which version of nRF52840 chip did you use for testing?

yuzhyang, 2018-04-17:
Thanks, @LuDuda. The hung node's chip version is Q1AAA 1646AD. More info for your reference, we have 1716AK and 1650AE in the testbed as well, which are not hung at present.

LuDuda, 2018-04-27:
Hello we have a suspect. We observed that this behavior is correlated with CSMA/CA algorithm.

And appears if this [function ](https://github.com/openthread/openthread/blob/master/third_party/NordicSemiconductor/drivers/radio/mac_features/nrf_802154_csma_ca.c#L134 ) draws 0 multiple times in a row.

 Unfortunately we are all ooo next week due to national holidays. We will try to fix it afterwards ASAP.

LuDuda, 2018-05-09:
Hello, I was able to reproduce failure behavior that was caused by race condition in radio.c.
The fix is on pull request. Is there any chance you can rerun your test on your end?

jwhui, 2018-05-15:
Resolved by #2700.

closed 2018-05-15 by jwhui