openthread/openthread #2665: nRF5 SDK for Thread example does not work with nrf_cc310 lib

kbogucki, 2018-04-16:
Hi,

I cannot run openthread example with change (add cryptocell support (#2628)):
https://github.com/openthread/openthread/commit/996be6e1474d405fdb10e83ce7d50c2a80fad6db

Example from nRF5 SDK for Thread (nRF5_SDK_for_Thread/examples/thread/experimental/freertos_coap_server) does not work when linking with openthread/third_party/NordicSemiconductor/libraries/nrf_cc310/lib/libnrf_cc310_0.9.9.a

The change I've made in example's Makefile:
removed lines:
`$(SDK_ROOT)/external/openthread/lib/gcc/libmbedcrypto-cc310.a \
$(SDK_ROOT)/external/nrf_cc310/lib/libcc310_gcc_0.9.0.a \
`

added lines:
`$(SDK_ROOT)/../openthread/build-node/nrf52840/third_party/mbedtls/libmbedcrypto.a \
$(SDK_ROOT)/../openthread/third_party/NordicSemiconductor/libraries/nrf_cc310/lib/libnrf_cc310_0.9.9.a \
`

The program hangs on call to `PlatformInit`

Any ideas how to make it working?

LuDuda, 2018-04-16:
Hello @kbogucki.

The new version of nRF5 SDK for Thread is going to be released in a few days maybe tomorrow) with your specific example fixed.

For the time being, can you try move all platform initialization out of FreeRTOS task, and call it from the main function, before FreeRTOS's thread is created?

kbogucki, 2018-04-17:
Hi @LuDuda,

Moving platform init to main function solved the problem.
Thanks for your help.
Waiting for new SDK.

Regards,
Krzysztof

closed 2018-04-17 by kbogucki