openthread/openthread #2682: Error compiling on Ubuntu 18.04

ekarlso, 2018-04-25:
latest trunk

./bootstrap
make JOINER=1 COMMISSIONER=1 USB=1 -f examples/Makefile-nrf52840 
  MKDIR    output
  MKDIR    build/nrf52840
  CONFIG   nrf52840...
(cd build/nrf52840 && /home/ekarlso/software/openthread/examples/../configure INSTALL="/usr/bin/install -p" CPP="arm-none-eabi-cpp" CC="arm-none-eabi-gcc" CXX="arm-none-eabi-g++" CCAS="arm-none-eabi-gcc" OBJC="cc" OBJCXX="" AR="arm-none-eabi-ar" RANLIB="arm-none-eabi-ranlib" NM="arm-none-eabi-nm" STRIP="arm-none-eabi-strip" CPPFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-nrf52840-config.h\"' -I/home/ekarlso/software/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -DUSB_CDC_AS_SERIAL_TRANSPORT=1 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs " CFLAGS=" -fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-nrf52840-config.h\"' -I/home/ekarlso/software/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -DUSB_CDC_AS_SERIAL_TRANSPORT=1 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs " CXXFLAGS=" -fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-nrf52840-config.h\"' -I/home/ekarlso/software/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -DUSB_CDC_AS_SERIAL_TRANSPORT=1 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs -fno-exceptions -fno-rtti " CCASFLAGS="  -x assembler-with-cpp " LDFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-nrf52840-config.h\"' -I/home/ekarlso/software/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -DUSB_CDC_AS_SERIAL_TRANSPORT=1 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs -specs=nano.specs -specs=nosys.specs -Wl,--gc-sections " --host=arm-none-eabi --prefix=/ --exec-prefix=/nrf52840 --enable-cli-app=all --enable-ncp-app=all --enable-diag --with-examples=nrf52840 --enable-linker-map MBEDTLS_CPPFLAGS="-DMBEDTLS_CONFIG_FILE='\"mbedtls-config.h\"' -DMBEDTLS_USER_CONFIG_FILE='\"nrf52840-mbedtls-config.h\"' -I/home/ekarlso/software/openthread/examples/../third_party/mbedtls -I/home/ekarlso/software/openthread/examples/../third_party/mbedtls/repo.patched/include -I/home/ekarlso/software/openthread/examples/../third_party/mbedtls/repo.patched/include/mbedtls -I/home/ekarlso/software/openthread/examples/../third_party/NordicSemiconductor/libraries/crypto -I/home/ekarlso/software/openthread/examples/../third_party/NordicSemiconductor/libraries/nrf_cc310/include -I/home/ekarlso/software/openthread/examples/../third_party/NordicSemiconductor/nrfx/mdk -I/home/ekarlso/software/openthread/examples/../third_party/NordicSemiconductor/cmsis -DNRF52840_XXAA"  --with-ncp-bus=uart --enable-commissioner --enable-joiner)
checking build system type... x86_64-unknown-linux-gnu
checking host system type... arm-none-eabi
checking filtered build system type... x86_64-unknown-linux-gnu
checking filtered host system type... arm-none-eabi
checking for a BSD-compatible install... /usr/bin/install -p
checking whether build environment is sane... yes
checking for arm-none-eabi-strip... arm-none-eabi-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether to disable executable checking... no
checking for style of include used by make... GNU
checking for arm-none-eabi-gcc... arm-none-eabi-gcc
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether arm-none-eabi-gcc accepts -g... yes
checking for arm-none-eabi-gcc option to accept ISO C89... none needed
checking whether arm-none-eabi-gcc understands -c and -o together... yes
checking dependency style of arm-none-eabi-gcc... gcc3
checking how to run the C preprocessor... arm-none-eabi-cpp
checking for arm-none-eabi-gcc... (cached) arm-none-eabi-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether arm-none-eabi-gcc accepts -g... (cached) yes
checking for arm-none-eabi-gcc option to accept ISO C89... (cached) none needed
checking whether arm-none-eabi-gcc understands -c and -o together... (cached) yes
checking dependency style of arm-none-eabi-gcc... (cached) gcc3
checking whether arm-none-eabi-gcc and cc understand -c and -o together... yes
checking whether we are using the GNU C++ compiler... yes
checking whether arm-none-eabi-g++ accepts -g... yes
checking dependency style of arm-none-eabi-g++... gcc3
checking how to run the C++ preprocessor... arm-none-eabi-g++ -E
checking whether we are using the GNU C++ compiler... (cached) yes
checking whether arm-none-eabi-g++ accepts -g... (cached) yes
checking dependency style of arm-none-eabi-g++... (cached) gcc3
checking whether arm-none-eabi-g++ understands -c and -o together... yes
checking dependency style of arm-none-eabi-gcc... gcc3
checking for arm-none-eabi-ar... arm-none-eabi-ar
checking for arm-none-eabi-ranlib... arm-none-eabi-ranlib
checking for arm-none-eabi-objcopy... arm-none-eabi-objcopy
checking for arm-none-eabi-strip... (cached) arm-none-eabi-strip
checking whether ln -s works... yes
checking for cmp... /usr/bin/cmp
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking whether arm-none-eabi-gcc is Clang... no
checking whether the C compiler understands -Wall... yes
checking whether the C compiler understands -Wextra... yes
checking whether the C compiler understands -Wshadow... yes
checking whether the C compiler understands -Werror... yes
checking whether the C compiler understands -std=c99... yes
checking whether the C compiler understands -pedantic-errors... yes
checking whether the C++ compiler understands -Wall... yes
checking whether the C++ compiler understands -Wextra... yes
checking whether the C++ compiler understands -Wshadow... yes
checking whether the C++ compiler understands -Werror... yes
[... 92 lines cut ...]
checking whether to enable DTLS due to joiner/commissioner... yes
checking whether to enable jam detection... no
checking whether to enable channel monitor... no
checking whether to enable channel manager... no
checking whether to enable mac filter... no
checking whether to enable diag... yes
checking whether to enable legacy... no
checking for NCP vendor hook source... none
checking for NCP Spinel Encrypter... none
no
checking whether to enable linker map output... yes
checking whether to enable examples... nrf52840
checking whether to build tools... yes
checking for doxygen... /usr/bin/doxygen
checking for dot... no
checking whether to build documentation... yes
configure: checking required package dependencies
checking for stdbool.h that conforms to C99... yes
checking for _Bool... yes
checking for ANSI C header files... (cached) yes
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking for strlcpy... configure: error: link tests are not allowed after AC_NO_EXECUTABLES
examples/Makefile-nrf52840:275: recipe for target 'build/nrf52840/config.status' failed
make: *** [build/nrf52840/config.status] Error 1

jwhui, 2018-04-25:
This error typically indicates that your arm gcc toolchain is not installed.

Does `which arm-none-eabi-gcc` return anything for you?

ekarlso, 2018-04-25:
which arm-none-eabi-gcc
/usr/bin/arm-none-eabi-gcc

Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/6.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:6.3.1+svn253039-1build1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 6.3.1 20170620 (15:6.3.1+svn253039-1build1)

jwhui, 2018-04-25:
It's not immediately obvious to me what is causing this issue.  It appears
that the link check against strlcpy is failing, which typically indicates
an issue with your build environment.

Can you provide your full config.log?

FWIW, master HEAD (2c4b793
<https://github.com/openthread/openthread/commit/2c4b7932835d3f155cbe3c885741d5ac5570255f>)
has passed Travis build checks with arm-none-eabi-gcc 6.3.1
<https://travis-ci.org/openthread/openthread/jobs/370756831>.

jwhui, 2018-04-26:
One thing to try is replicating the same steps that Travis uses to install gcc-arm-none-eabi.

https://github.com/openthread/openthread/blob/master/.travis/before_install.sh#L77

ekarlso, 2018-04-27:
Yeah, it seems it doesnt like GCC 6 .

closed 2018-04-27 by ekarlso

jwhui, 2018-04-27:
As mentioned above, Travis has build checks against GCC 6 for both x86 and arm, which are passing.