openthread/openthread #2686: [efr32] no uart output on ThunderBoard Sense 2 board

chshu, 2018-04-27:
There is no uart output when running latest OT build on [ThunderBoard Sense 2](https://www.silabs.com/products/development-tools/thunderboard/thunderboard-sense-two-kit) board, while uart input works well.

More details:

- Firmware: ot-cli-ftd built by`make -f examples/Makefile-efr32 BOARD=BRD4166A`
- Based on some debugging, the uart input works well, the device could receive cli command correctly.
- There is `transmitDone` callback for each `UARTDRV_Transmit()`, but no uart output on host side, so the issue seems in `UARTDRV_Transmit()` API.
- No this issue on [Wireless Starter Kit](https://www.silabs.com/products/development-tools/wireless/mesh-networking/mighty-gecko-starter-kit) board.

jwhui, 2018-04-27:
I confirm the same issue.  It has something to do with the UART flow control.  Applying this diff fixes the problem for me.

```c
diff --git a/examples/platforms/efr32/uart.c b/examples/platforms/efr32/uart.c
index cb00e194f..91fccd3fa 100644
--- a/examples/platforms/efr32/uart.c
+++ b/examples/platforms/efr32/uart.c
@@ -59,7 +59,7 @@ enum
             (USART_Parity_TypeDef)USART_FRAME_PARITY_NONE,    /* Parity */                       \
             (USART_OVS_TypeDef)USART_CTRL_OVS_X16,            /* Oversampling mode*/             \
             false,                                            /* Majority vote disable */        \
-            uartdrvFlowControlHwUart,                         /* Flow control */                 \
+            uartdrvFlowControlNone,                           /* Flow control */                 \
             BSP_SERIAL_APP_CTS_PORT,                          /* CTS port number */              \
             BSP_SERIAL_APP_CTS_PIN,                           /* CTS pin number */               \
             BSP_SERIAL_APP_RTS_PORT,                          /* RTS port number */              \
```

chshu, 2018-04-28:
@jwhui , The change also works for me, thanks:+1

nigelg-silabs, 2019-06-06:
Hi @jwhui, @chshu Can this be closed?

NikhilSarnaik, 2019-06-27:
This issue is present on the latest OT pull. I cannot see any cli output from the Thunderboard Sense 2. I tried to use the no-flow-control which did not fix it. Do we have any workarounds for this?

jwhui, 2019-07-01:
@NikhilSarnaik, I just tested on ThunderBoard Sense 2 with the latest master (ce26bff) and the workaround above worked for me.

NikhilSarnaik, 2019-07-02:
Hi @jwhui. I can confirm that the cli app works with the workaround on Thunderboard Sense 2 for me as well now. I had to reset the EFR to see the cli output. This is what I had missed before asking the question here, Thanks for the information.

jwhui, 2019-07-02:
Resolved by #3922.

closed 2019-07-02 by jwhui