openthread/openthread #2711: Alpine Linux & Musl libc compatibility: _IOC_SIZEBITS undefined

rdewaele, 2018-05-15:
I am experimenting with OpenThread in Docker using Alpine Linux, which uses musl libc. Compiling the unmodified sources resulted in an error about _IOC_SIZEBITS not being defined. Googling this error led me to the following page and prompted my solution included below: https://www.spinics.net/lists/linux-spi/msg08503.html

```
--- spi-hdlc-adapter.c    2018-05-06 16:36:19.000000000 +0200
+++ spi-hdlc-adapter.c  2018-05-15 19:14:09.000000000 +0200
@@ -50,6 +50,7 @@
 #include <sys/ioctl.h>
 #include <sys/file.h>

+#include <linux/ioctl.h>
 #include <linux/spi/spidev.h>

 #if HAVE_EXECINFO_H
```

I'm open to submitting this change as a PR, but maybe it's simpler for someone with pushing rights to just make this commit. Either way is fine.

jwhui, 2018-05-15:
@rdewaele, thanks for reporting this issue.  Please go ahead and submit a PR.

darconeous, 2018-05-15:
Good find! Happy to take this as a pull request.

PR #2712 (2018-05-17): [spi-hdlc-adapter] fix compilation on Alpine Linux with musl libc
Without linux/ioctl.h compilation fails on this platform, because _IOC_SIZEBITS will not be defined.

resolves #2711
closed 2018-05-17 by jwhui