openthread/openthread #2717: some commands not available (CoAP, Commissioner...)

BJeanjean, 2018-05-16:
Good Evening,

I'm using the EFR32™ Mighty Gecko Wireless Starter Kit, and I followed each step from this page https://github.com/openthread/openthread/tree/master/examples/platforms/efr32 .
However, I now have got some issue, in fact I would use CoAP and Commissioner commands for my project, but they are not available when I enter help into the console:

`help
autostart
bufferinfo
channel
child
childmax
childtimeout
contextreusedelay
counter
dataset
delaytimermin
diag
discover
eidcache
eui64
extaddr
extpanid
factoryreset
ifconfig
ipaddr
ipmaddr
joinerport
keysequence
leaderdata
leaderpartitionid
leaderweight
masterkey
mode
neighbor
networkdiagnostic
networkidtimeout
networkname
panid
parent
parentpriority
ping
pollperiod
promiscuous
pskc
releaserouterid
reset
rloc16
router
routerdowngradethreshold
routerrole
routerselectionjitter
routerupgradethreshold
scan
singleton
state
thread
txpower
udp
version
> 
`
And when I want to use those command, I have those messages:

`> commissioner start
Error 6: Parse`

`> coap start
Error 6: Parse`

I precise I'm using a virtual box Ubuntu 16.04, and I have enabled those features before compiling:
`$ make -f examples/Makefile-efr32 COMMISSIONER=1 COAP=1 JOINER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 BOARD=BRD4161A`

Thank you a lot for your anwsers,

Regards,

Bastien Jeanjean

jwhui, 2018-05-17:
@BJeanjean, thanks for reporting this issue.  I just tested with the latest master and was unable to reproduce your issue.

If you did a build without those optional features enabled, you may need to run configure again.  You could try running clean first and then rebuilding:
```bash
$ make -f examples/Makefile-efr32 BOARD=BRD4161A clean
$ make -f examples/Makefile-efr32 COMMISSIONER=1 COAP=1 JOINER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 BOARD=BRD4161A
```

Alternatively, you could try building from a clean repo:
```bash
$ git clean -xfd
$ git checkout -- .
$ ./bootstrap
$ make -f examples/Makefile-efr32 COMMISSIONER=1 COAP=1 JOINER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 BOARD=BRD4161A
```

You should see configuration output similar to:
```
  OpenThread CLI support                    : yes
  OpenThread CLI-MTD support                : yes
  OpenThread CLI-FTD support                : yes
  OpenThread NCP support                    : yes
  OpenThread NCP-MTD support                : yes
  OpenThread NCP-FTD support                : yes
  OpenThread NCP-BUS Configuration          : uart
  OpenThread NCP Vendor Hook Source         : none
  OpenThread NCP Spinel Encrypter           : none
  OpenThread Multiple Instances support     : no
  OpenThread MTD Network Diagnostic support : no
  OpenThread builtin mbedtls support        : yes
  OpenThread TMF Proxy support              : no
  OpenThread Commissioner support           : yes
  OpenThread Joiner support                 : yes
  OpenThread DTLS support                   : yes
  OpenThread Jam Detection support          : no
  OpenThread Channel Monitor support        : no
  OpenThread Channel Manager support        : no
  OpenThread MAC Filter support             : no
  OpenThread Diagnostics support            : yes
  OpenThread Child Supervision support      : no
  OpenThread Legacy network support         : no
  OpenThread Certification log support      : no
  OpenThread DHCPv6 Server support          : yes
  OpenThread DHCPv6 Client support          : yes
  OpenThread DNS Client support             : no
  OpenThread Application CoAP support       : yes
  OpenThread Raw Link-Layer support         : no
  OpenThread Border Router support          : no
  OpenThread Service support                : no
  OpenThread examples                       : efr32
```

Let me know if that helps.

BJeanjean, 2018-05-17:
Thanks for your fast answer.

I've try clean first before rebuilding and that now works with the leader. However I still cannot use CoAP with the Router, but I don't even know if I am supposed to be able to do so.
In fact, in my project I need one leader, and other nodes which could communicate in mesh Network, like we already can do with the examples from SiLabs (client/server), and I would use the CoAP method.

Thank you a lot.

jwhui, 2018-05-17:
@BJeanjean, did you try running through the [CoAP CLI example](https://github.com/openthread/openthread/blob/master/src/cli/README.md#coap-start)?  You should be able to use the CoAP service on any Thread device role (leader, router, child).  I just tried with the latest master (b76f72e) and it worked fine for me.

If you are still running into issues, can you provide more detail on the steps you performed so that we can reproduce the issue?

BJeanjean, 2018-05-18:
I retried to do everything again (even delete de gecko_sdk_suite folder to copy another one) and it seems to work now. I suppose that after cleaning I did something wrong when I flashed the second card.
Well thank you a lot for your help, and good luck for the future of OpenThread !

jwhui, 2018-05-25:
@BJeanjean, great to hear, and thanks!

closed 2018-05-25 by jwhui