openthread/openthread #2724: [posix] Dockerize OpenThread

rdewaele, 2018-05-18:
related: #1784, #2711 
possibly related: #1860 

I currently have a semi working version of OpenThread in Docker. The setup is to have a container running (Alpine Linux) for each OpenThread instance.
As the original POSIX platform "broadcasts" messages by looping over all possible ports, for each sending packets to 127.0.0.1, I modified this to use a multicast address. Each instance subscribes to this address but still has its own port, so I think this is a relatively small change.

The thing holding me back from making a separate PR is that for some reason all my OpenThread instances end up having the same mac address, resulting in them all having the same IPv6 address, which surely is hampering the correct operation. This is however most likely caused by other changes I made to the example code, to suit the specific use-cases I am exploring.

While I keep looking in to this:
- Is OpenThread interested in this approach for dockerization?
- Would OpenThread be interested in having the multicast change in a separate, independent PR?
- If someone happens to have a pointer for me where to start looking for the mac address thing, that would probably speed things along, because I'm a bit lost at the time of writing. ;-)

Nopik, 2018-10-31:
This is so much needed. Especially, if toranj tests could be run in docker as well, the setup would be way easier.

jwhui, 2018-11-27:
We now have a number of Docker images available:

1. [Simulating a Thread Network using OpenThread in Docker](https://codelabs.developers.google.com/codelabs/openthread-simulation/#0)
2. [Building OpenThread with a pre-configured dev environment](https://openthread.io/guides/build)
3. [OpenThread Border Router](https://openthread.io/guides/border-router/docker)

Hope that helps!

closed 2018-11-27 by jwhui