openthread/openthread #2726: multicast communication with Group ID supporting

wanglun2012, 2018-05-21:
Hello,
Does openthread support multicast communication like ZigBee?  
It needs a 16-bit network address called multicast address(Group ID).
Multicast: This is a transmission to every device in a particular PAN belonging to a dynamically defined multicast group, and within a given transmission radius measured in hops.

Thank you.

chshu, 2018-05-21:
Yes, OpenThread supports multicast, please refer to [Thread Spec 1.1.1](https://www.threadgroup.org/ThreadSpec) section 5.2.3.

By default, all Thread devices will subscribe to some Link-Local and Realm-Local multicast addresses (e.g., ff02::1, ff03::1).

You can also define you own multicast group, and subscribe your devices to it. Please refer to the [multicast related APIs](https://github.com/openthread/openthread/blob/master/include/openthread/ip6.h#L132).

wanglun2012, 2018-05-22:
@chshu 
Thanks for you reply firstly.
Thread Spec 1.1.1 section 5.2.3 just indicates supporting IPv6 multicast.
But it is different from ZigBee multicast extremely.

So, openthread just supports IPv6 multicast. Am I right?

chshu, 2018-05-22:
@wanglun2012 Yes, Thread is IPv6 based protocol.

wanglun2012, 2018-05-23:
@chshu 
I read the API which you mentioned.
But I don't know how to verify Openthread multicast.
For example, there are 4 nodes and I let 3 nodes subscribe multicast address ff05::1.
What should I do next?

chshu, 2018-05-23:
@wanglun2012 If you are using CLI version, you can simply use the ping cmd to verify:

Subscribe multicast address with cmd:
`ipmaddr add ff05::1`

Ping the multicast address from another node with cmd:
`ping ff05::1`

wanglun2012, 2018-05-23:
@chshu 
Thanks for you reply.
That's exactly what I were thinking.
I have already verified the multicast by ping cmd.

If I want to group control some lamps for demo show, maybe I should do some application development because Openthread lacks of higher application layer.

jwhui, 2018-05-25:
Yes, Thread provides IPv6 connectivity and is intentionally application-layer agnostic.

closed 2018-05-25 by jwhui

abtink, 2018-07-27:
@bokenator, multicast addresses can be added (subscribed) or removed using wpantund `"IPv6:MulticastAddresses"` property with `insertprop` (alias: `add`) or `removeprop` (alias: `remove`) commands, e.g.,  `wpanctl add IPv6:MulticastAddresses <address>`. `wpantund` will add the address on both NCP and the Linux network interface (`wpan<x>`).