openthread/openthread #2732: [test-diag] unit test linker error on OS X (10.13.4)

rdewaele, 2018-05-23:
## Platform Information

```
  Configuration Summary
  ---------------------
  Package                                   : OPENTHREAD
  Version                                   : 20170716-00602-g52644515-dirty
  Interface                                 : 0.0.5
  Build system                              : x86_64-apple-darwin17.5.0
  Host system                               : x86_64-apple-darwin17.5.0
  Host architecture                         : x86_64
  Host OS                                   : darwin17.5.0
  Cross compiling                           : no
  Build shared libraries                    : yes
  Build static libraries                    : yes
  Build debug libraries                     : no
  Build optimized libraries                 : yes
  Build coverage libraries                  : no
  Build coverage reports                    : no
  Address sanitizer support                 : no
  Lcov                                      : -
  Genhtml                                   : -
  Build tests                               : yes
  Build fuzz targets                        : no
  Build tools                               : yes
  OpenThread tests                          :
  Prefix                                    : /
  Documentation support                     : no
  Doxygen                                   : -
  GraphViz dot                              : -
  C Preprocessor                            : gcc -E
  C Compiler                                : gcc
  C++ Preprocessor                          : g++ -E
  C++ Compiler                              : g++
  Assembler Compiler                        : gcc
  Archiver                                  : ar
  Archive Indexer                           : ranlib
  Symbol Stripper                           : strip
  Object Copier                             :
  C Preprocessor flags                      : -O1 -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-posix-config.h"' -I/private/tmp/openthread/examples/../examples/platforms/posix/   -I${abs_top_srcdir}/third_party/mbedtls -I${abs_top_srcdir}/third_party/mbedtls/repo.patched/include -I${abs_top_srcdir}/third_party/mbedtls/repo.patched/include/mbedtls -DMBEDTLS_CONFIG_FILE=\"mbedtls-config.h\"
  C Compile flags                           :  -O1 -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-posix-config.h"' -I/private/tmp/openthread/examples/../examples/platforms/posix/   -Wall -Wextra -Wshadow -Werror -std=c99 -pedantic-errors -Wconversion -D_BSD_SOURCE=1 -D_DEFAULT_SOURCE=1
  C++ Compile flags                         :  -O1 -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-posix-config.h"' -I/private/tmp/openthread/examples/../examples/platforms/posix/   -Wall -Wextra -Wshadow -Werror -std=gnu++98 -Wno-c++14-compat -Wconversion -D_BSD_SOURCE=1 -D_DEFAULT_SOURCE=1
  Assembler flags                           :  -O1 -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-posix-config.h"' -I/private/tmp/openthread/examples/../examples/platforms/posix/
  Link flags                                : -O1 -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-posix-config.h"' -I/private/tmp/openthread/examples/../examples/platforms/posix/
  Link libraries                            :
  Link maps                                 : no
  Pretty                                    : ${abs_top_srcdir}/script/clang-format.sh
  Pretty args                               : -style=file -i
  Pretty check                              : ${abs_top_srcdir}/script/clang-format-check.sh
  Pretty check args                         : -
  OpenThread CLI support                    : yes
  OpenThread CLI-MTD support                : yes
  OpenThread CLI-FTD support                : yes
  OpenThread NCP support                    : yes
  OpenThread NCP-MTD support                : yes
  OpenThread NCP-FTD support                : yes
  OpenThread NCP-BUS Configuration          : uart
  OpenThread NCP Vendor Hook Source         : none
  OpenThread NCP Spinel Encrypter           : none
  OpenThread Multiple Instances support     : no
  OpenThread MTD Network Diagnostic support : yes
  OpenThread builtin mbedtls support        : yes
  OpenThread TMF Proxy support              : yes
  OpenThread Commissioner support           : yes
  OpenThread Joiner support                 : yes
  OpenThread DTLS support                   : yes
  OpenThread Jam Detection support          : yes
  OpenThread Channel Monitor support        : no
  OpenThread Channel Manager support        : no
  OpenThread MAC Filter support             : yes
[... 194 lines cut ...]
```

Using this compiler switch also solves the linker error:

```
❯ CFLAGS=-fno-common CXXFLAGS=-fno-common CPPFLAGS=-fno-common make -j8 -f examples/Makefile-posix
<snip>
❯ nm build/x86_64-apple-darwin/examples/platforms/posix/libopenthread-posix.a | grep _gPlatformPseudoResetWasRequested
000000000000112c S _gPlatformPseudoResetWasRequested
                 U _gPlatformPseudoResetWasRequested
❯ cd build/x86_64-apple-darwin
build/x86_64-apple-darwin❯ make check
<snip>
  CXX      test_diag.o
  CXXLD    test-diag
  CXX      test_ncp_buffer.o
  CXXLD    test-ncp-buffer
/Applications/Xcode.app/Contents/Developer/usr/bin/make  check-TESTS
PASS: test-aes
<snip>
```

So I think that if global variables _must_ be used, direct initialization will make sure they will not have common linkage, while also making sure that they are always properly initialized. Adding `-fno-common` or `-Wl,-load_all` (or equivalent) moves the fix to the build system and is compatible with the former.

P.S.: While I have found a number of solutions to make the problem go away, I am still not sure _why_ the linker refuses to load the misc.o object file all by itself and am really curious if this is normal behavior. :)

jwhui, 2018-05-30:
@rdewaele, thank you for raising this issue and detailed description!  If I read this correctly, your suggestion is to initialize globals at declaration.  If so, I agree and please submit a PR with the fix.

rdewaele, 2018-05-31:
@jwhui Indeed, initialising the globals fixed the issue in my testing. The alternative/complementary fix is to add some flags to the build system (compiler `-fno-common` and/or linker `-Wl,-load_all`) but these are more heavy handed I feel, and may have side effects.

Would you like me to scan the whole project for uninitialised globals, or change the one I experienced this problem with.

jwhui, 2018-05-31:
@rdewaele, I agree with not relying on compiler flags to address this. If you are willing, it would be great if you addressed all uninitialized globals.

rdewaele, 2018-06-03:
@jwhui Just a heads up, it's likely I'll only have time in the weekends for a bit to work on this, so it may take a week or two before I get around to wrap this up.

jwhui, 2018-07-09:
@rdewaele, any update on your progress?

rdewaele, 2018-07-09:
@jwhui I took some time off this and next weekend so I'll try and look into this either this or next Friday. Our project is currently focusing on parts different than OpenThread so that's why I didn't find the time yet.

bukepo, 2018-09-16:
Close this issue for #3049?

jwhui, 2018-09-27:
Resolved by #3049.

closed 2018-09-27 by jwhui