openthread/openthread #2733: CoAP ack does not reaching sender

NickNamerr, 2018-05-23:
Hello,

Just figured out that node does not recieved ACK message for CON CoAP Request.
What i made:
coap start
coap resource ipaddr
coap post 2001:db8:935::1 ipaddr con fd11:1111:1122:0:4e8f:f214:76d:f7a5

2001:db8:935::1 is eth0 of OTBR, there is running CoAP server and sending back message to ip from CoAP message

meanwhile ran tcpdump to see raw data on wpan0 interface. 
1) message from node
 fd11:22::6c3:4b8:ab7d:a8ef.5683 > 2001:db8:935::1.5683 

2) test message from OTBR to node (received successfully)
 fd11:1111:1122:0:c2d8:7b47:3180:28e9.45714 > fd11:1111:1122:0:4e8f:f214:76d:f7a5.5683

3) ACK message for 1 packet
fd11:22::70df:19ef:857c:86f1.5683 > fd11:22::6c3:4b8:ab7d:a8ef.5683

and node log:
Error receiving coap response message: Error 28: ResponseTimeout

ICMPv6 is reaching node:
ping6 fd11:22::6c3:4b8:ab7d:a8ef
PING fd11:22::6c3:4b8:ab7d:a8ef(fd11:22::6c3:4b8:ab7d:a8ef) 56 data bytes
64 bytes from fd11:22::6c3:4b8:ab7d:a8ef: icmp_seq=1 ttl=64 time=47.7 ms
64 bytes from fd11:22::6c3:4b8:ab7d:a8ef: icmp_seq=2 ttl=64 time=36.9 ms

Does CoAP service on node listen all interfaces?

jwhui, 2018-05-25:
@NickNamerr, there seems to be an issue with your CoAP server running on the OTBR.  The CoAP client is rejecting the ACK because the ACK message's IPv6 source address does not match the request message's IPv6 destination address.

RFC 7252 Section 4.1 states:

> With no security, the endpoint is solely identified by an
   IP address and a UDP port number.

RFC 7252 Section 5.3.2 states:

> The source endpoint of the response MUST be the same as the
       destination endpoint of the original request.

Can you check your CoAP server implementation?

jwhui, 2018-06-06:
Closing due to inactivity.

closed 2018-06-06 by jwhui

NickNamerr, 2018-09-26:
How to reopen this issue?

jwhui, 2018-09-26:
Please create a new issue.