openthread/openthread #2748: [mle_router]

LuDuda, 2018-05-25:
During testing of ot-ncp-radio example, i got this assert on posix side:

`src/core/thread/mle_router.cpp:1270: otError ot::Mle::MleRouter::HandleAdvertisement(const ot::Message&, const ot::Ip6::MessageInfo&): Assertion "leader != NULL" failed.`

Unfortunatelly i couldn't reproduce it to pull more debug information.

Thoughts?

LuDuda, 2018-05-30:
I got this assert again, but only once (it's very hard to reproduce), it is related with re-synchronization.

```
# Leader

> panid 0
Done
> ifconfig up
Done
> thread start
Done
> state
Leader
Done

# Child

> panid 0
Done
> ifconfig up
Done
> routerrole disable
Done
> thread start
Done

--- Reset of Child ---

> ifconfig up
Done
> thread start
Done

-- Wait for MLE Advertisement --

ASSERT
```

GDB Logs:

```
(gdb) where
#0  0x0002b390 in ot::Mle::MleRouter::HandleAdvertisement (this=0x20005dd8 <ot::sInstanceRaw+18192>, aMessage=..., aMessageInfo=...)
    at ../../../../openthread/src/core/thread/mle_router.cpp:1270
#1  0x00025296 in ot::Mle::Mle::HandleAdvertisement (this=0x20005dd8 <ot::sInstanceRaw+18192>, aMessage=..., aMessageInfo=...)
    at ../../../../openthread/src/core/thread/mle.cpp:2541
#2  0x00024fdc in ot::Mle::Mle::HandleUdpReceive (this=0x20005dd8 <ot::sInstanceRaw+18192>, aMessage=..., aMessageInfo=...)
    at ../../../../openthread/src/core/thread/mle.cpp:2452
#3  0x000248e8 in ot::Mle::Mle::HandleUdpReceive (aContext=0x20005dd8 <ot::sInstanceRaw+18192>, aMessage=0x20007998 <ot::sInstanceRaw+25296>, aMessageInfo=0x2003fe3c)
    at ../../../../openthread/src/core/thread/mle.cpp:2265
#4  0x0003dac0 in ot::Ip6::UdpSocket::HandleUdpReceive (this=0x20005e90 <ot::sInstanceRaw+18376>, aMessage=..., aMessageInfo=...)
    at ../../../../openthread/src/core/net/udp6.hpp:160
#5  0x0003e28c in ot::Ip6::Udp::HandleMessage (this=0x20005748 <ot::sInstanceRaw+16512>, aMessage=..., aMessageInfo=...)
    at ../../../../openthread/src/core/net/udp6.cpp:276
#6  0x00012fc4 in ot::Ip6::Ip6::HandlePayload (this=0x20005714 <ot::sInstanceRaw+16460>, aMessage=..., aMessageInfo=..., aIpProto=17 '\021')
    at ../../../../openthread/src/core/net/ip6.cpp:588
#7  0x0001345a in ot::Ip6::Ip6::HandleDatagram (this=0x20005714 <ot::sInstanceRaw+16460>, aMessage=..., aNetif=0x200057f8 <ot::sInstanceRaw+16688>, aInterfaceId=1 '\001',
    aLinkMessageInfo=0x2003ff14, aFromNcpHost=false) at ../../../../openthread/src/core/net/ip6.cpp:820
#8  0x000519f4 in ot::MeshForwarder::HandleDatagram (this=0x20005d18 <ot::sInstanceRaw+18000>, aMessage=..., aLinkInfo=..., aMacSource=...)
    at ../../../../openthread/src/core/thread/mesh_forwarder.cpp:1416
#9  0x00051968 in ot::MeshForwarder::HandleLowpanHC (this=0x20005d18 <ot::sInstanceRaw+18000>, aFrame=0x2000c053 <nrf_802154_rx_buffers+26> "", aFrameLength=42 '*',
    aMacSource=..., aMacDest=..., aLinkInfo=...) at ../../../../openthread/src/core/thread/mesh_forwarder.cpp:1394
#10 0x00051288 in ot::MeshForwarder::HandleReceivedFrame (this=0x20005d18 <ot::sInstanceRaw+18000>, aFrame=...)
    at ../../../../openthread/src/core/thread/mesh_forwarder.cpp:1139
#11 0x000510c4 in ot::MeshForwarder::HandleReceivedFrame (aReceiver=..., aFrame=...) at ../../../../openthread/src/core/thread/mesh_forwarder.cpp:1092
#12 0x00018778 in ot::Mac::Receiver::HandleReceivedFrame (this=0x20005d18 <ot::sInstanceRaw+18000>, aFrame=...) at ../../../../openthread/src/core/mac/mac.hpp:321
#13 0x0001af8a in ot::Mac::Mac::HandleReceivedFrame (this=0x20005ab8 <ot::sInstanceRaw+17392>, aFrame=0x2000bb10 <sReceivedFrames>, aError=OT_ERROR_NONE)
    at ../../../../openthread/src/core/mac/mac.cpp:2025
#14 0x0001aab0 in ot::Mac::otPlatRadioReceiveDone (aInstance=0x200016c8 <ot::sInstanceRaw>, aFrame=0x2000bb10 <sReceivedFrames>, aError=OT_ERROR_NONE)
    at ../../../../openthread/src/core/mac/mac.cpp:1761
#15 0x0005a09c in nrf5RadioProcess (aInstance=aInstance@entry=0x200016c8 <ot::sInstanceRaw>) at ../../../../../openthread/examples/platforms/nrf52840/radio.c:536
#16 0x00059c7e in PlatformProcessDrivers (aInstance=0x200016c8 <ot::sInstanceRaw>) at ../../../../../openthread/examples/platforms/nrf52840/platform.c:119
#17 0x000066bc in thread_process () at ../../../../../../../components/thread/utils/thread_utils.c:106
#18 0x00006072 in main (argc=0, argv=0x0 <__isr_vector>) at ../../../../main.c:125

(gdb) p mRouterTable
$2 = {
  <ot::InstanceLocator> = {<No data fields>},
  members of ot::RouterTable:
  mRouters = {{
      <ot::Neighbor> = {
        mMacAddr = {
          <otExtAddress> = {
            m8 = "\000\000\000\000\000\000\000"
          }, <No data fields>},
        mLastHeard = 0,
        mValidPending = {
          mValid = {
            mLinkFrameCounter = 0,
            mMleFrameCounter = 0
          },
          mPending = {
            mChallenge = "\000\000\000\000\000\000\000"
          }
        },
        mKeySequence = 0,
        mRloc16 = 65535,
        mState = 0 '\000',
        mMode = 0 '\000',
        mDataRequest = false,
        mLinkFailures = 0 '\000',
        mLinkInfo = {
          mRssAverager = {
            mAverage = 0,
            mCount = 0
          },
          mLinkQuality = 0 '\000',
          mLastRss = 0 '\000',
          mFrameErrorRate = {
            mFailureRate = 0
          },
          mMessageErrorRate = {
            mFailureRate = 0
          }
        }
      },
      members of ot::Router:
---Type <return> to continue, or q <return> to quit---
      mNextHop = 0 '\000',
      mLinkQualityOut = 0 '\000',
      mCost = 0 '\000'
    } <repeats 32 times>},
  mAllocatedRouterIds = "\000\000\000\000\000\000\000",
  mRouterIdReuseDelay = '\000' <repeats 62 times>,
  mRouterIdSequenceLastUpdated = 0,
  mRouterIdSequence = 123 '{',
  mActiveRouterCount = 0 '\000'
}

(gdb) p mLeaderData
$4 = {
  <ot::Mle::Tlv> = {
    <ot::Tlv> = {
      mType = 0 '\000',
      mLength = 0 '\000'
    }, <No data fields>},
  members of ot::Mle::LeaderDataTlv:
  mPartitionId = 1690959718,
  mWeighting = 64 '@',
  mDataVersion = 0 '\000',
  mStableDataVersion = 0 '\000',
  mLeaderRouterId = 27 '\033'
}

```

It seems like mLeaderData has Leader Id set to 27 but mRouterTable is empty. When i put breakpoint in this line, it looks like in correct situation mRouterRable contains Leader object.

LuDuda, 2018-05-30:
@jwhui i think i found the reason, the problem was introduced in #2678.

The assert happens after re-synchronization of FTD child if this condition fails: https://github.com/openthread/openthread/blob/master/src/core/thread/mle_router.cpp#L1195

Because of that we don't call `ProcessRouteTlv` method and so `mRouterTable` stays empty.

The assert is hard to reproduce since it depends on the random value of `mRouterTable.mRouterIdSequence`
https://github.com/openthread/openthread/blob/master/src/core/thread/router_table.cpp#L83

I believe this bug was also present before but it didn't cause any assert. We just overwrited some random element of `mRouters` array.

What is your suggestion about the fix?

I did quick a look on other places, and in couple of seconds i was able to get assert also here:
https://github.com/openthread/openthread/blob/master/src/core/thread/mle_router.cpp#L1526

```
# Leader

> panid 0
Done
> ifconfig up
Done
> thread start
Done
> state
Leader
Done

# Child 1

> panid 0
Done
> ifconfig up
Done
> thread start
Done
> state
Child
Done
> reset

# Child 2

> panid 0
Done
> ifconfig up
Done

# Child 1 (before Leader sends MLE Advertisement)

> ifconfig up
Done
> thread start
Done

# Child 2 (before Leader sends MLE Advertisement)

> thread start
Done

# Child 1
src/core/thread/mle_router.cpp:1527: otError ot::Mle::MleRouter::HandleParentRequest(const ot::Message&, const ot::Ip6::MessageInfo&): Assertion `leader != NULL' failed.
```

jwhui, 2018-05-30:
@LuDuda, thanks for identifying this issue!  At first glance, it seems we need a couple fixes:

1. In `MleRouter::HandleAdvertisement()`, force processing the Route64 TLV if the router table is not initialized yet (i.e. after first attaching to the network).
2. In `MleRouter::HandleParentRequest()`, replace the assert with `VerifyOrExit()`.

I'll think more about the how best to resolve this issue.  Let me know if you have any further thoughts.

jwhui, 2018-05-31:
A proposed fix is in #2759.  Please review.

PR #2759 (2018-06-04): [mle] properly handle cases where router table is not populated
After attaching to a network, a device may need to wait until receiving the
next MLE Advertisement before populating the router table.

This commit makes the following changes:

- When receiving an MLE Advertisement, always process the Route TLV when the
  router table is not yet populated.

- Have RouterTable::GetLeaderAge() return max value when the router table
  is not yet populated.

- When generating a Connectivity TLV, populate the max routing cost when
  no leader information is available.

Resolves #2748.
closed 2018-06-04 by jwhui