openthread/openthread #2762: Reassembly timeout while joining to network with preconfigured node

VictorAtPL, 2018-06-03:
I am trying to connect to working network with the CLI node.

The attach process doesn't want to finish. There are repetitions of following frames/packets:
[image]

In node which I am trying to connect I can see following logging info:
> [INFO]-MAC-----: Dropping (reassembly queue) IPv6 UDP msg, len:197, chksum:2d3a, sec:yes, error:ReassemblyTimeout, prio:low, rs

Please take a look at sniffer packets whether whole tranmission looks good to you. I'd like to know whether the bug seems to be in joining node or leader node.

Log from node, which tries to connect: [unsucessful-join.log](https://github.com/openthread/openthread/files/2066195/unsucessful-join.log)
PCAP-NG file from Wireshark: [unsucessful-join.zip](https://github.com/openthread/openthread/files/2066197/unsucessful-join.zip)

jwhui, 2018-06-05:
@VictorAtPL, AFAICT, it seems that the the attaching child is not properly acknowledging the Child ID Response fragments.

Note that the zip file you attached seems to contain the log, rather than the pcap.

Also, can you provide more details on the platform(s) you are using?

VictorAtPL, 2018-06-05:
@jwhui I am sorry, I've uploaded .log file twice rather than pcap-ng file in zip. Now it's changed. Take a look at first post.

I am using TI CC 2538. Not sure what more do you want to know. 

Why could it not properly send ACK messages? Timing? Doesn't it look like a bug in OpenThread?

I've also tried this patch: https://github.com/openthread/openthread/pull/2371 and it didn't help.

jwhui, 2018-06-06:
@VictorAtPL, I was not successful at reproducing this issuing using the latest commit (3eb552a).

I set up a Thread Leader using a CC2538.  Then with another CC2538 in factory-reset state, I performed the following:
```
> mode rs
Done
> panid 0x1234
Done
> ifconfig up
Done
> thread start
Done
> state
child
Done
```

Can you provide detailed steps to help us reproduce the problem?

VictorAtPL, 2018-06-09:
I can not reproduce it now, too.
I am closing the issue. I will reopen it if it occurs again.

closed 2018-06-09 by VictorAtPL