openthread/openthread #2781: Openthread platform pin/peripheral configuration is hardcoded

nealjack, 2018-06-12:
For all platforms, the UART pin/peripheral configurations are hardcoded in the Openthread library. For example  `openthread/examples/platforms/kw41z/uart.c` hardcodes the use of UART0 and the UART pins as set as PTC6 and PTC7. This makes it difficult to use the library on any board besides the development kits these choices were based on. Many modern SoCs allow extensive pin/peripheral muxing to allow flexible design, and many projects won't use the same pin configuration as the library currently supports.

Pin/peripheral configuration should be defined in a separate board file that allows customization for various projects.

jwhui, 2018-06-12:
Thanks for suggesting this enhancement.

Note that the example drivers are not a part of the OpenThread library and are only intended to provide example implementations of the platform APIs defined in [include/openthread/platform](https://github.com/openthread/openthread/tree/master/include/openthread/platform).  It is not a goal of the OpenThread project to deliver a fully-featured SDK across the example platforms.

That said, we welcome and encourage any efforts to make the example drivers support different board configurations (e.g. #2566 and #2763).  If the example drivers can leverage vendor-supplied SDKs to support various board configurations, even better!

nealjack, 2018-06-12:
When the openthread libraries are complied, are they compiled as platform agnostic binaries?

My previous experience was with the Nordic Thread SDK, where it contained a precompiled binary that seemed to have pins for the dev kits hardcoded into it. I might be mistaken or remembering wrong, and I haven't tried the new SDK, but if this is the current norm for compiling the library it should be changed to not include the layers below the HAL.

closed 2018-07-09 by jwhui