openthread/openthread #2798: [ip6] memory leak when passing frame from NCP to Host

LuDuda, 2018-06-15:
Hello,

**Background:**
Recently I have found a serious issue while performing OpenThread NCP stress tests.

I have 24x NCP devices (nRF52840) in my setup that flood network (approx. for ~8h) with Echo Request packets on FF03::1 with different sizes. I control devices using pyspinel wpan API.

The result i found is that after test (or in the middle of the test) i see a flood of _MLE Parent Request_ messages. The end result is that all devices become Thread Leaders.

I saw also some posts from other users that discovered this problem with a 2-3 devices in the network.

**Investigation:**
I simply rebuilt firmware without optimization and put breakpoint in the `MleRouter::HandleParentRequest`, then i found that `MleRouter::SendParentResponse` fails because there is no free buffers (even couple of hours after performing stress tests).

To find a memory leak i have added profiling method to `Message` class e.g:

```
VerifyOrExit((message = NewMleMessage()) != NULL, error = OT_ERROR_NO_BUFS);

#if OPENTHREAD_CONFIG_MESSAGE_PROFILING
    message->SetConsumerFuncName(__PRETTY_FUNCTION__);
#endif
```

Also i set `ConsumerFuncName` to NULL in `MessagePool::Free` method.

_Side Note:  I don't feel this profiling method is the greatest, though i haven't find better way. Anyway, do you think that this kind of profiling has potential to be a part of OpenThread?_

After problem occurred again i printed logs from the GDB, and voilà, a lot of buffers pointed to `Ip6::ProcessReceiveCallback`:

```
      mHead = {
            mInfo = {
              mNext = {0x2000486c <ot::sInstanceRaw+15964>, 0x2000486c <ot::sInstanceRaw+15964>},
              mPrev = {0x200030ec <ot::sInstanceRaw+9948>, 0x200030ec <ot::sInstanceRaw+9948>},
              mMessagePool = 0x20002a68 <ot::sInstanceRaw+8280>,
              mQueue = {
                mMessage = 0x200000f8 <ot::Ncp::sNcpRaw+32>,
                mPriority = 0x200000f8 <ot::Ncp::sNcpRaw+32>
              },
              mReserved = 0,
              mLength = 84,
              mOffset = 40,
              [...]
              mTxSuccess = false,
              mConsumerFuncName = 0x5a9e0 <ot::Ip6::Ip6::ProcessReceiveCallback(ot::Message const&, ot::Ip6::MessageInfo const&, unsigned char, bool)::__PRETTY_FUNCTION__> "otError ot::Ip6::Ip6::ProcessReceiveCallback(const ot::Message&, const ot::Ip6::MessageInfo&, uint8_t, bool)"
            },
           [...]
```

Anyway the strange thing is that when i use `otMessageGetBufferInfo` i see that all buffers all free?!

I am not sure yet where the problem is, i have two candidates and i am going to check them, however i would appreciate if you @jwhui can look on this as well.

LuDuda, 2018-06-15:
Update.. It turns out that the problem is related to the test procedure. After the test i did close all Spinel connections, and because of that buffers start to fill NCP queues and eventually we end up with no free buffers. Do you see any solution for that, since i still consider this as unwanted behaviour?

Anyway during investigation i found that memory leak may actually happen if https://github.com/openthread/openthread/blob/master/src/ncp/ncp_base_mtd.cpp#L2859 fails.
Also we don't check error codes e.g. here: https://github.com/openthread/openthread/blob/master/src/ncp/ncp_base_mtd.cpp#L2924

I guess because we clone the new buffer above scenarios are not likely to occur, but if you think we should add additional checks, i can send PR with them.

abtink, 2018-06-15:
@LuDuda, I am curious to understand what caused the leak. 

About the `ncp_base_mdt.cpp` code, `otMessageQueueEnqueue` or `otMessageQueueEnqueueAtHead` return the following errors
```
 * @retval OT_ERROR_NONE     Successfully added the message to the queue.
 * @retval OT_ERROR_ALREADY  The message is already enqueued in a queue.
```
In both cases, the message is included in the queue (it's owned by the `mMessageQueue` queue). Then later the message is dequeued (`otMessageQueueDequeue()`)  and passed to `SendDatagramMessage()`
- In case of success, it is owned by the NCP buffer (check `SpinelEncoder::WriteMessage`), and would be freed when the spinel frame is sent (frame is removed from NCP buffer).
- In case of failure (out of NCP spinel buffer), the message is put back into the queue  using `otMessageQueueEnqueueAtHead()` which should not fail (since we just dequeued it from same queue so it cannot be in the queue)

LuDuda, 2018-06-15:
Hello @abtink as i wrote in last comment, I discovered that the problem was not related with the memory leak. When i turned off host side, message queue in the ncp_base_mtd has started to fill in, and eventually i believe all buffers were placed in this queue. Because of that OT was unable to allocate message to reply with MLE Parent Response etc. Do you think it's correct behaviour, in such cases - Host goes offline?

abtink, 2018-06-15:
Thanks @LuDuda. It makes sense.
We keep the messages in NCP spinel buffer until we can deliver them to host. I think this is the desired behavior for NCP build. NCP build relies on the host. Host can go to sleep/low-power mode and then we have `otPlatWakeHost()` mechanism and `SPINEL_PROP_HOST_POWER_STATE ` from OpenThread to wake host (when there is a message for host).

LuDuda, 2018-06-17:
Thank you @abtink for clarification. If this is desired behavior for NCP i am going to close this issue.

As a future improvement, i think it would be good to leave some buffers (e.g. 10% of all) for the stack to allow mesh operations such as packet forwarding or prevent situations where we end up in Leader role of a new network - as described in my first post.

closed 2018-06-17 by LuDuda