openthread/openthread #2799: OpenThread cannot work after using link raw API

KaSroka, 2018-06-15:
When after using link raw API we want to start Thread network it won't work correctly as radio would be left in disabled state.

To reproduce (see comments inline):
```c
otError error;

error = otThreadSetEnabled(thread_ot_instance_get(), false);
ASSERT(error == OT_ERROR_NONE);

error = otIp6SetEnabled(thread_ot_instance_get(), false);
ASSERT(error == OT_ERROR_NONE);

error = otLinkRawSetEnable(thread_ot_instance_get(), true); // This call will try to enable already enabled radio - on some implementations it may fail
ASSERT(error == OT_ERROR_NONE);

error = otLinkRawSetEnable(thread_ot_instance_get(), false); // Radio will be disabled after this call 
ASSERT(error == OT_ERROR_NONE);

error = otIp6SetEnabled(thread_ot_instance_get(), true);  // Radio is enabled only in Mac constructor so it will remain disabled
ASSERT(error == OT_ERROR_NONE);

error = otThreadSetEnabled(thread_ot_instance_get(), true);
ASSERT(error == OT_ERROR_NONE);
```

To make it work we need to add call to `otPlatRadioEnable` before we enable Ip6/Thread and this should be avoided.

jwhui, 2018-07-09:
@KaSroka, thanks for pointing this out.  Are you willing to submit a PR to address this issue?

jwhui, 2023-06-27:
Closing stale issue.

closed 2023-06-27 by jwhui