openthread/openthread #2802: Minor code questions

Nopik, 2018-06-16:
During OpenThread code analysis I've stumbled into few bits of code which generated some questions. I'm opening this issue as an umbrella one to host them all, to avoid issue spamming with single sentence stuff. If it will be necessary, I'll split separate issues and/or PRs.

Here we go:

- there are two `RouteTlv` structures, inheriting over two different bases, apart from it they are pretty identical. But one has `uint8_t mRouteData[kMaxRouterId + 1]` and `uint8_t mRouterIdMask[BitVectorBytes(kMaxRouterId + 1)]` and the other is missing `+1`. Is that correct?

- this `if` statement seems to be always true: https://github.com/openthread/openthread/blob/3eb552a9b7ee7a0dae65e61b4fa6d6b631daa4a4/src/core/thread/router_table.cpp#L537 and therefore can be simplified.

- `https://github.com/openthread/openthread/blob/12237d07dc813f72a16d4d6d4681e75680dd7569/src/core/thread/network_diagnostic_tlvs.hpp#L630` and `https://github.com/openthread/openthread/blob/51c62a29d0ce170e8c85e65f9f8deade0d7c575e/src/core/thread/mle_tlvs.hpp#L547` seem to be unused. Are they left there for the sake of completeness (their `set` counterpart is there and is being used) or should be removed?

jwhui, 2018-06-18:
Thanks for pointing these out.

> - there are two RouteTlv structures, inheriting over two different bases, apart from it they are pretty identical. But one has uint8_t mRouteData[kMaxRouterId + 1] and uint8_t mRouterIdMask[BitVectorBytes(kMaxRouterId + 1)] and the other is missing +1. Is that correct?

`NetworkDiagnostic::RouteTlv::mRouterIdMask` should add the `+ 1` for consistency.  Although, the `+ 1` in this specific case does not change anything.

`NetworkDiagnostic::RouteTlv::mRouteData` should change from `Mle::kMaxRouters` to `Mle::kMaxRouterId + 1` fo consistency as well.

> - this `if` statement seems to be always true: https://github.com/openthread/openthread/blob/3eb552a9b7ee7a0dae65e61b4fa6d6b631daa4a4/src/core/thread/router_table.cpp#L537 and therefore can be simplified.

Yes, because of the `aTlv.IsRouterIdSet(i) == IsAllocated(i)` check above, I believe you are correct and this could be simplified.

> - `https://github.com/openthread/openthread/blob/12237d07dc813f72a16d4d6d4681e75680dd7569/src/core/thread/network_diagnostic_tlvs.hpp#L630` and [...]

I believe those were included for consistency.  However, because they are unused, I am fine with removing them.

Feel free to submit PR(s) to implement the above changes.

Nopik, 2018-06-18:
Thanks, PR submitted.

closed 2018-06-18 by Nopik