openthread/openthread #2803: Path cost calculation errors

Nopik, 2018-06-16:
During my work on #2775 I've been testing how path cost is being calculated, and I've noticed some oddities. Maybe it is me not understanding the fine details of the Thread spec, but more likely this is some bug. Here it goes:

First, I've been testing 'long line' kind of topology: devices are in line, each device can see only 1 neighbour each side, end ones can see only 1 neighbour. Pretty much identical to #2782. Each link was having RSSI set to `-90`, which should correspond to route cost of `4` on each hop.

I have WIP implementation of 'long routes' mode, where I can form a line of >4 routers. For starting, I've been playing with 8 of them. When I collected all their path costs taken from `router table` output, the table was mostly OK, but with 2 strange behaviours:

from \ to | dc00 | 8c00 | bc00 | 5c00 | 3c00 | 7800 | b000 | 1000
--------- | ---- | ---- | ---- | ---- | ---- | ---- | ---- | ----
dc00 | 0 | *0* | 4 | 8 | 12 | 16 | 20 | 24
8c00 | 8 | 0 | 8 | 4 | 8 | 12 | 16 | 20
bc00 | 4 | 8 | 0 | 8 | 4 | 8 | 12 | 16
5c00 | 8 | 4 | 8 | 0 | 8 | 4 | 8 | 12
3c00 | 12 | 8 | 4 | 8 | 0 | 8 | 4 | 8
7800 | 16 | 12 | 8 | 4 | 8 | 0 | 8 | 4
b000 | 20 | 16 | 12 | 8 | 4 | 8 | 0 | 8
1000 | 24 | 20 | 16 | 12 | 8 | 4 | *0* | 0

How to read a table: row header is source rloc16, column header is destination rloc16. IDs are arranged in topology order, so rloc `dc00` was at one end of the line, while `1000` was the other end. So, `route table` output for `dc00` shown that it thinks that the cost to `8c00` is `0`, whereas `8c00` thinks that its cost to `dc00` is `8`.

As you see following 2 problems are striking:
- typically each node think that the cost to closest neighbour is `8`, the cost to one after that is `4`, and then all other costs are too small by `4`. I.e. I would expect cost from `dc00` to `1000` to be `28` (7 hops) not `24`.
- cost from end nodes to their closest neighbours is `0` for some reason. It is probably related to the 'next hop' value from `router table` being `63`.

Is that how it is supposed to work? I've expected pretty much `0 - 4 - 8 - ... - 24 - 28` kind of output.

So, I suspected, that my PR had broken something. I did a clean checkout of current master, and re-tried with 4 nodes. Here is the table:

to \ from | dc00 | 8c00 | bc00 | 5c00
--------- | ---- | ---- | ---- | ----
dc00 | 0 | *0* | 4 | 8
8c00 | 8 | 0 | 8 | 4
bc00 | 4 | 8 | 0 | 8
5c00 | 8 | 4 | *0* | 0

Which is consistent with the larger table.

Interestingly, lack of `3c00` node changed cost from `5c00` to `bc00` from `8` to `0`.

Also, if the max cost is 16, I should be able to add 1 more node and see 12 in the table, right?

After a bit closer digging, I've also noticed that 'next hop' value get weirdly reported. On each node I did `router 0x5c00`, and node `bc00` said that the next hop is `8c00`, whereas it should be directly `5c00`. 

Another oddity is that the LQ In / LQ Out values in table, sometimes shown `1` instead of `0`. But, there was no `2` or `6` in the route cost table...

Also, I waited 5 real-time minutes (nodes were run with `20` time speedup factor) before issuing `router table` command, to make sure that the routing costs stabilized properly.

Here is the cli dump of all 4 nodes:

```
Worker 1
> channel 11
Done
> panid 0x1234
Done
> ifconfig up
Done
> routerselectionjitter 1
Done
> extaddr
720fbb94bbdfd3b8
Done
> macfilter addr whitelist
Done
> macfilter addr clear
Done
> macfilter rss clear
Done
> macfilter addr add e2de40e9ab097f9a
Done
> macfilter rss add e2de40e9ab097f9a -90
Done
> thread start
Done
> state
leader
Done

< Wait until all devices are routers>

> rloc16
dc00
Done
> router 0x5c00
Alloc: 1
Router ID: 23
Rloc: 5c00
Next Hop: 8c00
Link: 0
Done
> coap start
Coap service started: Done
> coap get fdde:ad00:beef:0:0:ff:fe00:5c00 test con
Sending coap request: Done
> Received coap response with payload: 30
state
leader
Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
| 23 | 0x5c00 |       35 |         8 |     0 |      0 |   9 | b6ef87b37a19d6a0 |
| 35 | 0x8c00 |       63 |         0 |     1 |      1 |  14 | e2de40e9ab097f9a |
| 47 | 0xbc00 |       35 |         4 |     0 |      0 |  13 | 2a24f980296c7f07 |
| 55 | 0xdc00 |       63 |         0 |     0 |      0 |   0 | 720fbb94bbdfd3b8 |
Done

Worker 2
> channel 11
Done
> panid 0x1234
Done
> ifconfig up
Done
> routerselectionjitter 1
Done
> extaddr
e2de40e9ab097f9a
Done
> macfilter addr whitelist
Done
> macfilter addr clear
Done
> macfilter rss clear
Done
> macfilter addr add 2a24f980296c7f07
Done
> macfilter rss add 2a24f980296c7f07 -90
Done
> macfilter addr add 720fbb94bbdfd3b8
Done
> macfilter rss add 720fbb94bbdfd3b8 -90
Done
> thread start
Done
> state
router
Done

< Wait until all devices are routers>

> rloc16
8c00
Done
> router 0x5c00
Alloc: 1
Router ID: 23
Rloc: 5c00
Next Hop: bc00
Link: 0
Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
| 23 | 0x5c00 |       47 |         4 |     0 |      0 |  45 | 0000000000000000 |
| 35 | 0x8c00 |       63 |         0 |     0 |      0 |   0 | e2de40e9ab097f9a |
| 47 | 0xbc00 |       55 |         8 |     1 |      1 |  16 | 2a24f980296c7f07 |
| 55 | 0xdc00 |       47 |         8 |     1 |      1 |   6 | 720fbb94bbdfd3b8 |
Done

Worker 3
> channel 11
Done
> panid 0x1234
Done
> ifconfig up
Done
> routerselectionjitter 1
Done
> extaddr
2a24f980296c7f07
Done
> macfilter addr whitelist
Done
> macfilter addr clear
Done
> macfilter rss clear
Done
> macfilter addr add b6ef87b37a19d6a0
Done
> macfilter rss add b6ef87b37a19d6a0 -90
Done
> macfilter addr add e2de40e9ab097f9a
Done
> macfilter rss add e2de40e9ab097f9a -90
Done
> thread start
Done

< Wait until all devices are routers>

> rloc16
bc00
Done
> state
router
Done
> router 0x5c00
Alloc: 1
Router ID: 23
Rloc: 5c00
Next Hop: 8c00
Link: 1
Ext Addr: b6ef87b37a19d6a0
Cost: 8
Link Quality In: 1
Link Quality Out: 1
Age: 6
Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
| 23 | 0x5c00 |       35 |         8 |     1 |      1 |   7 | b6ef87b37a19d6a0 |
| 35 | 0x8c00 |       23 |         8 |     1 |      1 |  14 | e2de40e9ab097f9a |
| 47 | 0xbc00 |       63 |         0 |     0 |      0 |   0 | 2a24f980296c7f07 |
| 55 | 0xdc00 |       35 |         4 |     0 |      0 |  45 | 0000000000000000 |
Done

Worker 4
> channel 11
Done
> panid 0x1234
Done
> ifconfig up
Done
> routerselectionjitter 1
Done
> extaddr
b6ef87b37a19d6a0
Done
> macfilter addr whitelist
Done
> macfilter addr clear
Done
> macfilter rss clear
Done
> macfilter addr add 2a24f980296c7f07
Done
> macfilter rss add 2a24f980296c7f07 -90
Done
> thread start
Done

< Wait until all devices are routers>

> state
router
Done
> coap start
Coap service started: Done
> coap resource test
Resource name is 'test': Done
> rloc16
5c00
Done
> coap start
Coap service started: Done
> Received coap request from [fdde:ad00:beef:0:0:ff:fe00:5c00]: GET
coap response sent successfully!
state
router
Done
> router 0x5c00
Alloc: 1
Router ID: 23
Rloc: 5c00
Next Hop: fc00
Link: 0
Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
| 23 | 0x5c00 |       63 |         0 |     0 |      0 |   0 | b6ef87b37a19d6a0 |
| 35 | 0x8c00 |       47 |         4 |     0 |      0 | 145 | 0000000000000000 |
| 47 | 0xbc00 |       63 |         0 |     1 |      1 |  18 | 2a24f980296c7f07 |
| 55 | 0xdc00 |       47 |         8 |     0 |      0 | 145 | 0000000000000000 |
Done
```

Can you elaborate on these findings, please? Are those bugs, or I'm missing something?

jwhui, 2018-06-16:
@Nopik, thank you for sharing your findings with such detail!

I think everything you have described is as designed and expected.

In particular, the route entries maintain both path cost and link cost separately.  Route cost is the best path to a given router ID that is advertised by a neighboring router.  In your 'long routes' example above, `8c00` -> `bc00` is `8` because it represents the path `8c00` -> `dc00` -> `8c00`  -> `bc00`.  Yes, the path is a loop, but the routing protocol has no way to determine that on its own.  For the case of `dc00` -> `8c00`, the path cost is 0 because there is no multihop path that exists (even with loops).

 Link cost is the cost to reach the neighbor directly.  Of course, the link cost itself is `4` and when forwarding a packet, the next hop lookup will compare communicating with the neighbor directly or route via a different neighbor.

Having the route entries separate link and path costs gives an opportunity to continue forwarding messages even if a direct link no longer exists.

You can find additional detail in #1187.

As for the link cost itself, only seeing value `4` is expected.  Your CLI output shows that links are configured to simulate -90 dBm RSSI.  The posix platform has sensitivity level hard-coded at -100 dBm.  As a result, the link margins are 10 dB, which corresponds to link quality `1` and link cost `4`, as defined in Thread 1.1.1 Section 5.9.4.

Hope that helps.

Nopik, 2018-06-16:
Thanks for clarification. Indeed, all the cases seem to be my wrong interpretation of the results, not bugs. I thought that the path cost displayed by `route table` is just a cost to reach given node, but apparently it is `best cost from neighbours to reach that node`, which basically explains everything.

And for the LQ stuff, for some reason I expected all the values to be the same - and since there were 0's in majority I assumed, they correspond to -90 dBm (knowing about 5.9.4 table 5-5, just not consulting the values with it). So when I saw some `1`s there, I assumed it is odd. On a closer look I've realized that `1` is exactly where I set the visibility to be on. Just a thinko, I guess ;)

closed 2018-06-16 by Nopik