openthread/openthread #2804: CLI output is truncated

Nopik, 2018-06-16:
When cli is trying to display large chunks of text, the output gets truncated. For example, when having 12 routers, the output of `router table` gets cut like this:

```
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
|  2 | 0x0800 |       35 |         1 |     3 |      3 |  12 | b6a448e0654eba9b |
|  8 | 0x2000 |       35 |         1 |     3 |      3 |  12 | 827401a6195f1ffc |
| 12 | 0x3000 |       35 |         1 |     3 |      3 |  11 | a2d514287266e554 |
| 25 | 0x6400 |       35 |         1 |     3 |      3 |  27 | bafa7f1b81c05b57 |
| 27 | 0x6c00 |       35 |         1 |     3 |      3 |  14 | a62742e5e1a1c342 |
| 29 | 0x7400 |       35 |         1 |     3 |      3 |  25 | eef033db43db7e02 |
| 32 | 0x8000 |       35 |         1 |     3 |      3 |  24 | e2fed024c8ed9e52 |
| 35 | 0x8c00 |        8 |         1 |     3 |      3 |  16 | be0d1b0a35acca16 |
| 42 | 0xa800 |       63 |         0 |     0 |      0 |   0 | 6683e2da94a42743 |
| 44 | 0xb000 |       35 |         1 |     3 |      3 |  20 | 1e9a5ae4f2a1cdf0 |
| 51 | 0xcc00 |       35 |         1 |
```

which seems to be at 1000 bytes mark or very close, at least on my platform (osx).

To reproduce, just launch 12 nodes, do this on each of it:
```
channel 11
panid 0x1234
ifconfig up
routerselectionjitter 1
thread start
```
wait for routers to appear and issue `router table` on any of the nodes.

LuDuda, 2018-06-17:
Hello @Nopik, CLI output is truncated because of maximum TX buffer size, set to 1024 by default, see `OPENTHREAD_CONFIG_CLI_UART_TX_BUFFER_SIZE` ([config](https://github.com/openthread/openthread/blob/master/src/core/openthread-core-default-config.h#L261)).

Nopik, 2018-06-17:
Thanks, I expected something like that, but didn't knew which value it is. I guess it is not a bug, then. I'll close this issue and if anyone thinks that it actually should be fixed (e.g. by bumping the bufsize to 2k), we can reopen.

closed 2018-06-17 by Nopik