openthread/openthread #2806: Create standalone ARM 32 bit library for Embedded System

raupune, 2018-06-18:
Dear, 
I have gone through example set given here. But these examples are limited while real world projects requirements are way different.

I want to create a standalone library file (.a extension for now) and use that file in my project.

I will write all necessary files required for PAL(platform abstraction layer) at my own. (For e.g. implementation of radio.c file).

But I am not able to find out instructions on how to create a standalone library file for embedded projects.

One pretty good implementation of such requirement is RIOT OS. They have almost platform independent implementation of OpenThread.

Please add such instructions on how to build standalone ARM 32 bit library file.

jwhui, 2018-06-18:
Each [example platform](https://github.com/openthread/openthread/tree/master/examples) has a README that walks through the build process.  The library files that you are looking for are generated as part of that process.  For example:
```
$ ls output/cc2538/lib/
libmbedcrypto.a			libopenthread-diag.a		libopenthread-ncp-mtd.a
libopenthread-cc2538.a		libopenthread-ftd.a		libopenthread-ncp-radio.a
libopenthread-cli-ftd.a		libopenthread-mtd.a		libopenthread-platform-utils.a
libopenthread-cli-mtd.a		libopenthread-ncp-ftd.a		libopenthread-radio.a
```
Where `libopenthread-ftd.a` is the OpenThread library for Full Thread Device (FTD).

Of course, feel free to integrate OpenThread into your own build system as well.

Hope that helps.

raupune, 2018-06-19:
Dear,

Thanks for prompt reply. 

I went through examples and its pretty good

But will you please elaborate how can I integrate Openthread in my own build system? 

Integrating Openthread Library in other projects build environment is a genuine requirement. Will you please address the problem?

I went through RIOT OS  Openthread implementation. They imported only required Openthread Stack. And they ask user to implement HAL ( Hardware Abstraction Layer). E.g. radio.c to be implemented by user.

I want to know how to do this? 

Consider this as a feature request to be added in build environment.

jwhui, 2018-06-19:
Integrating OpenThread into your project is largely dependent on your project-specific toolchain and development flows.  While the OpenThread repo utilize the GNU toolchain, OpenThread is designed to be toolchain agnostic.

In addition to RIOT-OS, some other projects that have integrated OpenThread are Zephyr and IoTivity-Lite.  Silicon vendors have also integrated OpenThread into their own vendor-specific SDKs, such as Nordic and Texas Instruments.

raupune, 2018-06-21:
Dear jwhui,
Thanks for prompt reply. 
My idea is to build a Device/Architecture independent 32 bit library. 
Put that library in another project. 
Write HAL ( Hardware Abstraction Layer) in respective project.
But question still remains, How to build such independent library?
Openthread says it's Device/Platform agnostic (hardware should be 802.11.15.4 compliant) . 
Then it should be possible to create an independent Library.
Kindly correct me if am wrong.
If yes then how to proceed? 
Thanks.

jwhui, 2018-06-21:
In your original post, you wanted to generate a static library (`.a`).  Static libraries are compiled for a specific instruction set.  So, by definition, the static libraries are architecture-dependent.

As mentioned in my previous post, cach [example platform](https://github.com/openthread/openthread/tree/master/examples) has a README that walks through the build process.  The library files that you are looking for are generated as part of that process.

For example, if you follow the build process for the CC2538, you will find the following libraries after a successful build:
```
$ ls output/cc2538/lib/
libmbedcrypto.a			libopenthread-diag.a		libopenthread-ncp-mtd.a
libopenthread-cc2538.a		libopenthread-ftd.a		libopenthread-ncp-radio.a
libopenthread-cli-ftd.a		libopenthread-mtd.a		libopenthread-platform-utils.a
libopenthread-cli-mtd.a		libopenthread-ncp-ftd.a		libopenthread-radio.a
```

Those libraries expect to link against a platform-specific library that implement the `otPlat*` APIs.

closed 2018-07-09 by jwhui

NOA-Vernon, 2019-10-24:
Dear,
I want to build the latest OpenThread libraries with GCC on Linux for nrf52840 SoC, and do the steps as the following link
https://infocenter.nordicsemi.com/topic/sdk_tz_v3.2.0/thread_ot_libraries_building.html?cp=5_3_2_1_4_0#thread_ot_libraries_building_1
however, the output libraries didn't have libopenthread-diag.a, see the screen shot below

[image]

how can I generate the  libopenthread-diag.a in the output/nrf52840/lib/ folder? thank you in advance.

NOA-Vernon, 2019-10-25:
@jwhui Thanks a lot for the clarification.