openthread/openthread #2812: Error During Make -f

BJeanjean, 2018-06-19:
Good Morning,

A few days ago, an update of the openthread has been released, the I decided to download the new version of it. I am currently working on the EFR32MG12P433F1024GM48 module on a custom device, I have updated all the SDK from Simplicity Studio, but when I want to do the "make -f", I have new errors I never had before:

````
make -f examples/Makefile-efr32 BOARD=BRD4161A 
  MKDIR    output
  MKDIR    build/efr32
  CONFIG   efr32...
(cd build/efr32 && /home/romain/Documents/OpenThread/openthread/examples/../configure INSTALL="/usr/bin/install -p" CPP="arm-none-eabi-cpp" CC="arm-none-eabi-gcc" CXX="arm-none-eabi-g++" OBJC="cc" OBJCXX="" AR="arm-none-eabi-ar" RANLIB="arm-none-eabi-ranlib" NM="arm-none-eabi-nm" STRIP="arm-none-eabi-strip" CPPFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-efr32-config.h\"' -I/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/ -I/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32//brd4161a -DEFR32MG12P432F1024GL125  -mcpu=cortex-m4 -mfloat-abi=soft -mthumb " CFLAGS=" -fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-efr32-config.h\"' -I/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/ -I/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32//brd4161a -DEFR32MG12P432F1024GL125  -mcpu=cortex-m4 -mfloat-abi=soft -mthumb " CXXFLAGS=" -fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-efr32-config.h\"' -I/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/ -I/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32//brd4161a -DEFR32MG12P432F1024GL125  -mcpu=cortex-m4 -mfloat-abi=soft -mthumb -fno-exceptions -fno-rtti " LDFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-efr32-config.h\"' -I/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/ -I/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32//brd4161a -DEFR32MG12P432F1024GL125  -mcpu=cortex-m4 -mfloat-abi=soft -mthumb -nostartfiles -specs=nano.specs -specs=nosys.specs -Wl,--gc-sections " --host=arm-none-eabi --prefix=/ --exec-prefix=/efr32 --enable-cli-app=all --enable-ncp-app=all --with-ncp-bus=uart --enable-diag --with-examples=efr32 --enable-linker-map MBEDTLS_CPPFLAGS="-DMBEDTLS_CONFIG_FILE='\"mbedtls-config.h\"' -DMBEDTLS_USER_CONFIG_FILE='\"efr32-mbedtls-config.h\"' -DEFR32MG12P432F1024GL125 -I/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/crypto -I/home/romain/Documents/OpenThread/openthread/examples/../third_party/silabs/gecko_sdk_suite/v2.2/util/third_party/mbedtls/configs -I/home/romain/Documents/OpenThread/openthread/examples/../third_party/silabs/gecko_sdk_suite/v2.2/platform/CMSIS/Include -I/home/romain/Documents/OpenThread/openthread/examples/../third_party/silabs/gecko_sdk_suite/v2.2/util/third_party/mbedtls/sl_crypto/include -I/home/romain/Documents/OpenThread/openthread/examples/../third_party/silabs/gecko_sdk_suite/v2.2/platform/Device/SiliconLabs/EFR32MG12P/Include -I/home/romain/Documents/OpenThread/openthread/examples/../third_party/silabs/gecko_sdk_suite/v2.2/platform/emlib/inc -I/home/romain/Documents/OpenThread/openthread/examples/../third_party/mbedtls -I/home/romain/Documents/OpenThread/openthread/examples/../third_party/mbedtls/repo/include -I/home/romain/Documents/OpenThread/openthread/examples/../third_party/mbedtls/repo/include/mbedtls" )
checking build system type... x86_64-unknown-linux-gnu
checking host system type... arm-none-eabi
checking filtered build system type... x86_64-unknown-linux-gnu
checking filtered host system type... arm-none-eabi
checking for a BSD-compatible install... /usr/bin/install -p
checking whether build environment is sane... yes
checking for arm-none-eabi-strip... arm-none-eabi-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether to disable executable checking... no
checking for style of include used by make... GNU
checking for arm-none-eabi-gcc... arm-none-eabi-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether arm-none-eabi-gcc accepts -g... yes
checking for arm-none-eabi-gcc option to accept ISO C89... none needed
checking whether arm-none-eabi-gcc understands -c and -o together... yes
checking dependency style of arm-none-eabi-gcc... gcc3
checking how to run the C preprocessor... arm-none-eabi-cpp
checking for arm-none-eabi-gcc... (cached) arm-none-eabi-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether arm-none-eabi-gcc accepts -g... (cached) yes
checking for arm-none-eabi-gcc option to accept ISO C89... (cached) none needed
checking whether arm-none-eabi-gcc understands -c and -o together... (cached) yes
checking dependency style of arm-none-eabi-gcc... (cached) gcc3
checking whether arm-none-eabi-gcc and cc understand -c and -o together... yes
checking whether we are using the GNU C++ compiler... yes
checking whether arm-none-eabi-g++ accepts -g... yes
checking dependency style of arm-none-eabi-g++... gcc3
checking how to run the C++ preprocessor... arm-none-eabi-g++ -E
checking whether we are using the GNU C++ compiler... (cached) yes
checking whether arm-none-eabi-g++ accepts -g... (cached) yes
checking dependency style of arm-none-eabi-g++... (cached) gcc3
checking whether arm-none-eabi-g++ understands -c and -o together... yes
checking dependency style of arm-none-eabi-gcc... gcc3
checking for arm-none-eabi-ar... arm-none-eabi-ar
checking for arm-none-eabi-ranlib... arm-none-eabi-ranlib
checking for arm-none-eabi-objcopy... arm-none-eabi-objcopy
checking for arm-none-eabi-strip... (cached) arm-none-eabi-strip
checking whether ln -s works... yes
checking for cmp... /usr/bin/cmp
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking whether arm-none-eabi-gcc is Clang... no
checking whether the C compiler understands -Wall... yes
checking whether the C compiler understands -Wextra... yes
checking whether the C compiler understands -Wshadow... yes
checking whether the C compiler understands -Werror... yes
checking whether the C compiler understands -std=c99... yes
[... 628 lines cut ...]
    ^~~~~
/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/../../../third_party/silabs/gecko_sdk_suite/v2.2/util/plugin/plugin-common/fem-control/fem-control.c: In function 'initFem':
/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/../../../third_party/silabs/gecko_sdk_suite/v2.2/util/plugin/plugin-common/fem-control/fem-control.c:116:19: error: 'BSP_FEM_SLEEP_PORT' undeclared (first use in this function); did you mean 'BSP_FEM_TX_PORT'?
   GPIO_PinModeSet(BSP_FEM_SLEEP_PORT, BSP_FEM_SLEEP_PIN, gpioModePushPull, 0);
                   ^~~~~~~~~~~~~~~~~~
                   BSP_FEM_TX_PORT
/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/../../../third_party/silabs/gecko_sdk_suite/v2.2/util/plugin/plugin-common/fem-control/fem-control.c:116:19: note: each undeclared identifier is reported only once for each function it appears in
/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/../../../third_party/silabs/gecko_sdk_suite/v2.2/util/plugin/plugin-common/fem-control/fem-control.c:116:39: error: 'BSP_FEM_SLEEP_PIN' undeclared (first use in this function); did you mean 'BSP_FEM_SLEEP_PORT'?
   GPIO_PinModeSet(BSP_FEM_SLEEP_PORT, BSP_FEM_SLEEP_PIN, gpioModePushPull, 0);
                                       ^~~~~~~~~~~~~~~~~
[... cut ...]
/home/romain/Documents/OpenThread/openthread/examples/../examples/platforms/efr32/../../../third_party/silabs/gecko_sdk_suite/v2.2/util/plugin/plugin-common/fem-control/fem-control.c:140:22: error: 'BSP_FEM_SLEEP_LOC' undeclared (first use in this function); did you mean 'BSP_FEM_SLEEP_PIN'?
                      BSP_FEM_SLEEP_LOC << ((BSP_FEM_SLEEP_CHANNEL % 4) * 8));
                      ^~~~~~~~~~~~~~~~~
                      BSP_FEM_SLEEP_PIN
```
`

I don't really know what to do, and I would like to know if this issue was already known and how to fix it.

Bastien J.

tomas-c, 2018-06-19:
Might be due to #2763. Looking into it now.

tomas-c, 2018-06-19:
Indeed. If in `./examples/platforms/efr32/Makefile.am` you remove `@top_builddir@/third_party/silabs/gecko_sdk_suite/v2.2/util/plugin/plugin-common/fem-control/fem-control.c` and rerun `./bootstrap` then it should fix the problem. I will create a pull request to fix this later.

tomas-c, 2018-06-19:
@jwhui Looks like we need a conditional compilation on that file. Some EFR32 boards need it and some don't. I'm thinking it could be done in two ways:

1. Define BOARD_INCLUDES and BOARD_SOURCES in ./examples/Makefile-efr32 and pass them to ./examples/platforms/efr32/Makefile.am.

2. Add conditionals to configure.ac, maybe something like OPENTHREAD_EXAMPLES_EFR32_<board_name> and pick the right files in ./examples/platforms/efr32/Makefile.am using those conditionals.

What do you think?

BJeanjean, 2018-06-20:
It worked ! Thank you a lot

jwhui, 2018-06-20:
@tomas-c, at this point I'm leaning towards adding the includes and sources to `examples/Makefile-efr32`.  Ideally, `configure.ac` is limited to building the core OpenThread library.  The example platforms are currently intended to be examples for demonstrating the OpenThread library.  Thoughts?

tomas-c, 2018-06-25:
@jwhui After more investigation, I could not see an obvious way to pass board-specific sources from <ot_root>/examples/Makefile-efr32 into <ot_root>/examples/platforms/efr32/Makefile.am so not sure about the first option anymore. I pushed #2830 as an alternative fix for this particular problem.

jwhui, 2018-06-26:
Resolved by #2830.

closed 2018-06-26 by jwhui